arduino/Arduino #9013: Unable to update installed libraries in Library Manager

per1234, 2019-06-24:
Arduino IDE 1.9.0-beta build 170
Windows 10 64 bit / Ubuntu 19.04

1. Tools > Manage Libraries...
2. Select an installed library that has other releases available.
3. From the "Select version" dropdown menu, select any version.
4. Click the "Install" button.

The process fails with the error:
```
 An error occurred while installing library!
java.lang.RuntimeException: java.lang.Exception: An error occurred while installing library!
	at cc.arduino.contributions.libraries.ui.LibraryManagerUI.lambda$onInstallPressed$3(LibraryManagerUI.java:224)
	at java.lang.Thread.run(Thread.java:748)
Caused by: java.lang.Exception: An error occurred while installing library!
	at cc.arduino.contributions.libraries.LibraryInstaller.install(LibraryInstaller.java:94)
	at cc.arduino.contributions.libraries.ui.LibraryManagerUI.lambda$onInstallPressed$3(LibraryManagerUI.java:216)
	... 1 more
```
The same error occurs after clicking the "Update" button.

There is no problem when clicking the "Install" button of a library that is not already installed.

The issue does not occur with Arduino IDE 1.8.10 Hourly Build 2019/05/21 09:33.

sinnfeinn, 2019-06-25:
same here:
Arduino beta 1.9.0
Ubuntu 18.0 LTS

An error occurred while installing library!
java.lang.RuntimeException: java.lang.Exception: An error occurred while installing library!
	at cc.arduino.contributions.libraries.ui.LibraryManagerUI.lambda$onInstallPressed$3(LibraryManagerUI.java:224)
	at java.lang.Thread.run(Thread.java:748)

facchinm, 2019-06-25:
In Beta channel Library Manager is handled by `arduino-cli` so the problem could be:
* the integration or
* the cli directly

Since the version of the cli that was "merged" is quite archaic it's probably an issue with it; to properly fix all the integration code should be converted to the grpc APIs and the cli updated.