arduino/Arduino #9014: Arduino ide doesnt download any packages

TomasRoj, 2019-06-24:
I tried almost every link for esp8266 package. Nothing works. Everything gives error with downloading. I hadnt any problem with downloading theese in previous version 1.8.8.

per1234, 2019-06-24:
>I tried almost every link for esp8266 package

This is the only Boards Manager URL you should ever use for ESP8266:
https://arduino.esp8266.com/stable/package_esp8266com_index.json
Does that URL cause an error? If so, post the full text of the error here.

TomasRoj, 2019-06-25:
> Komu: arduino/Arduino <Arduino@noreply.github.com>
> 9014)
> ""

Yes still the same. It gives following a error : Nepodařilo se stáhnout 
balíček + link. In English : can't download package +link. 

https://arduino.esp8266.com/stable/package_esp8266com_index.json
(https://arduino.esp8266.com/stable/package_esp8266com_index.json)

(https://github.com/arduino/Arduino/issues/9014?email_source=notifications&email_token=AKDLLF7FE4LXNBLXXUBHWLLP4FB3PA5CNFSM4H26L6E2YY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGODYOMQEY#issuecomment-505202707)
"

per1234, 2019-06-25:
Where you say "+ link", is that really what the error says, or did you just leave that part of the error out?

After you get that error and click the "Close" button on Boards Manager, do you see some additional error output in the black console window at the bottom of the Arduino IDE window? If so, please post the full text of that output here. You might need to scroll the console up to see it all.

TomasRoj, 2019-06-25:
yes, I just skipped the link addres and not it doesnt print more errors

TomasRoj, 2019-06-27:
Any suggests?

closed 2021-04-04 by per1234