arduino/Arduino #9015: TinyStepper_28BYJ_48 :No such file or directory

shihaipeng03, 2019-06-24:
Although there is TinyStepper_28BYJ_48 in LIB, it cannot be compiled correctly.  An error will be reported. In file included from /tmp/948630601/Walldraw/Walldraw.ino:9:0:

/home/builder/opt/libraries/latest/tinystepper_28byj_48_1_0_0/src/TinyStepper_28BYJ_48.h:14:21: fatal error: arduino.h: No such file or directory

per1234, 2019-06-24:
This is caused by a bug in the TinyStepper_28BYJ_48 library, which I have now submitted a pull request to fix: https://github.com/Stan-Reifel/TinyStepper_28BYJ_48/pull/1

Until the library author merges that PR and makes a new release, the workaround is to import the fixed version of the library:

1. Download https://github.com/per1234/TinyStepper_28BYJ_48/archive/fix-include-case.zip
2. (In the Arduino Web Editor) Click the upward pointing arrow button (import).
3. If you get a message about importing your sketchbook, click the "Import" button.
4. Select the downloaded file.
5. Click the "Open" button.
6. Wait until the Arduino Web Editor tells you the library was successfully imported. It will now compile.

closed 2019-06-24 by per1234

shihaipeng03, 2019-06-25:
> now submitted a pull request to fix: Stan-Reifel/TinyStepper_28BYJ_48#1

Thank you. Follow your approach. It is now ready to use.