arduino/Arduino #9017: Misleading clock speed on Nano Every product page

per1234, 2019-06-25:
https://store.arduino.cc/nano-every

The "Tech Specs" tab of the Nano Every product page says:
>Clock 	20MHz

but the Arduino megaAVR Boards core is intentionally configured to run this board at 16 MHz:
https://github.com/arduino/ArduinoCore-megaavr/blob/master/boards.txt#L56

Although the board is capable of running at 20 MHz, to do so would require the user to modify the Arduino megaAVR Boards core, which is a fairly advanced process. For this reason, I feel the 20 MHz clock speed stated on the product page is a bit misleading and should be changed to:
>Clock 	16 MHz (20 MHz max)

swilkens, 2020-07-14:
At the very least, a straightforward method to employ 20 Mhz should be provided.

holgerl, 2023-01-04:
What is happening in this case? It still states 20 MHz on the product page, but in practice only 16 MHz is enabled