arduino/Arduino #9022: Arduino Leonardo only working on Windows 10 pc, its not working on Linux

ajay1234444, 2019-06-28:
when I compile a sketch in Arduino Leanardo using Linux it shows an error but using windows it works without any problem

error in Linux:
    avrdude: verification error, first mismatch at byte 0x0000 arduino leonardo

PaulStoffregen, 2019-06-28:
Check if ModemManager is interfering

ajay1234444, 2019-06-28:
every other board are working fine with Linux.

matthijskooijman, 2019-06-28:
> every other board are working fine with Linux.

But most other boards work differently from the Leonard, which has native USB and really needs access to the serial device shortly after the serial port is (re)added, which is exactly what ModemManager prevents. Also, in recent versions (Debian buster and Ubuntu Disco IIRC), ModemManager has disabled its blacklists and started probing Arduino boards again. See https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=930264

So, I also strongly suggest you disable MM (or even better, uninstall, since MM sometimes autorestarts) and see if that helps.

Having said that: The error message does not look like MM interfering, since the upload protocol works fine, but somehow the data read back is incorrect.

Do you have a second Leonardo (or Micro) board to rule out problems with this particular board? Does the problem happen with a blank sketch as well? What happens if you press the reset button to force bootloader mode (the led will slowly fade) and then start the upload?

ajay1234444, 2019-06-28:
I  Arduino Leonardo I can only upload the basic blink sketch using Arduino IDE in Linux also the same problem with the Arduino pro micro clone .when I upload  sketch except for the blink sketch the error happens

ajay1234444, 2019-06-28:
after shutting down the ModemManager I upload the blink sketch to an Arduino pro micro 

error :
 Found programmer: Id = "
CATERI"; type = S

avrdude: butterfly_recv(): programmer is not responding
avrdude: Expected signature for ATmega32U4 is 1E 95 87
         Double check chip, or use -F to override this check.
Wrong microcontroller found.  Did you select the right board from the Tools > Board menu?

matthijskooijman, 2019-06-28:
I'm confused. First you seem to be saying you *can* upload the blink sketch but not others, then you say the blink sketch isn't working with an error message? Can you provide a complete (but also as small as possible) sketch which is failing to upload?

Also, your error message seems truncated, could you make sure to paste more complete messages? Best to use ``` before and after the pasted text, so it is formatted as pre-formatted text.

Note that when you upload a sketch that behaves badly (e.g. disables interrupts for long periods of time or otherwise interferes with the USB communication), it might be that the *next* sketch you try to upload fails. Perhaps this is what your seeing and the problem isn't in the sketch you're trying to upload, but in the previous sketch. In all cases, you should be able to work around this by pressing the reset button and make sure avrdude starts uploading while the led is still fading.

ajay1234444, 2019-06-28:
actually, every sketch uploads fine with windows 10 in Arduino Leanardo and Arduino pro micro. But when I use Linux to upload the sketch to both boards first time it uploads the blink sketch without any problem I upload any other sketch other than the blink sketch (for example the standardFirmata sketch) the Arduino IDE show the above error

ajay1234444, 2019-06-28:
I solve the problem by stopping  ModemManager in Linux but sometimes it ended up with a new error

error:
  avrdude: error: buffered memory access not supported. Maybe it isn't
a butterfly/AVR109 but a AVR910 device?
avrdude: initialization failed, rc=-1
         Double check connections and try again, or use -F to override
         this check.

avrdude: error: programmer did not respond to command: leave prog mode
avrdude: error: programmer did not respond to command: leave prog mode

matthijskooijman, 2019-06-29:
Did you try another USB cable? There was someone in the #arduino chat channel today who also had the "buffered memory access not supported." message and solved it by changing the USB cable...

ajay1234444, 2019-06-29:
sudo apt-get purge modemmanager solve the problem

ajay1234444, 2019-06-30:
I found a new error while uploading this  sketch to Arduino Leanardo  on Linux 

error :
     avrdude: error: buffered memory access not supported. Maybe it isn't
a butterfly/AVR109 but a AVR910 device?
avrdude: initialization failed, rc=-1
         Double check connections and try again, or use -F to override
         this check.

the selected serial port 
 does not exist or your board is not connected

this is the Arduino sketch I used:
[nunchuck_funcs .h.txt](https://github.com/arduino/Arduino/files/3342545/nunchuck_funcs.h.txt)

[wiimouse(ino file ).txt](https://github.com/arduino/Arduino/files/3342543/wiimouse.ino.file.txt)

Aeduino IDE Version :1.8.9
linux version : Release 4.7 64-bit
                      Kernel Linux 4.19.37-parrot1-amd64 x86_64
                      MATE 1.20.4