arduino/Arduino #9024: Arduino Nano Every Analog Pins Issues

seisfeld, 2019-06-28:
Hi,

I'm using IDE 1.8.9. on macOS and megaAVR 1.8.1 is installed.

It appears there is an issue with the new megaAVR based Arduino Nano Every and its analog pins. This code does not blink an LED connected to pin A0, instead the LED stays off. In contrast it works fine when when using LED_BUILTIN. This has been verified on multiple brand new boards.

```c++
void setup() {
  pinMode(A0, OUTPUT);
}

void loop() {
  digitalWrite(A0, HIGH);
  delay(1000); 
  digitalWrite(A0, LOW);
  delay(1000);
}
```

Similar, input on the analog pins also does not seem to work. The following code does not indicate 0 when A0 is pulled to GND. Instead it always shows 1 (in contrast digital pin 10 works fine):

```c++
uint8_t testPin(A10);

void setup()
{
    Serial.begin(115200);
    Serial.println("Start");
    pinMode(testPin, INPUT_PULLUP);
}

void loop()
{
    Serial.print(millis());
    Serial.print(' ');
    Serial.println(digitalRead(testPin));
    delay(1000);
}
```

Is this a general issue or am I overseeing something very obvious?

Thanks
Stephan

facchinm, 2019-06-28:
Hi @seisfeld ,
the bug was discovered just after releasing 1.8.1 and already fixed in master (https://github.com/arduino/ArduinoCore-megaavr/commit/4e03d352e4d310d9fd6ebec5fb4c8225fe5e4c4a) .

I'm going to publish 1.8.2 early next week; in the meantime, if you want to test the fix, I'm attaching a core you can use to replace the existing one
[megaavr-1.8.2-beta.zip](https://github.com/arduino/Arduino/files/3339868/megaavr-1.8.2-beta.zip)

seisfeld, 2019-06-28:
Hey @facchinm,

thanks for your prompt reply. I unzipped the 1.8.2-beta, copied it in place and retried both examples above. Both now work as expected! Awesome!

Thanks again for your assistance. I'm looking forward to the next release. :)

cheers
Stephan

facchinm, 2019-07-01:
I just published official 1.8.2, closing the issue as fixed :wink:

closed 2019-07-01 by facchinm