arduino/Arduino #9034: Add drwattson (UpbeatLabs_MCP39F521) to Library Manager

sridharrajagopal, 2019-07-01:
Please add the Arduino library from drwattson (https://github.com/upbeatlabs/drwattson) to the Library Manager. The Arduino library can be found at https://github.com/upbeatlabs/drwattson/tree/master/Arduino/UpbeatLabs_MCP39F521 and the repository has been tagged. 

UpbeatLabs_MCP39F521 is an Arduino library to interface withe Microchip MCP39F521 to get energy data (Vrms, Irms, Power Factor, Active Power, Reactive Power etc) easily to an Arduino project. 

Thanks,
Sridhar

per1234, 2019-07-02:
In order to be added to the Arduino Library Manager index, the Arduino library must be located in the root of the repository, rather than in a subfolder as it is now.

I would suggest moving the Arduino library to its own dedicated repository, since the users of the Arduino library and the users of Python library are two distinct groups. However, that's just my take on things. You are free to find whatever solution you think best that puts the library in the root of the repository.

Once you have done this, create a release or tag and leave a comment here so we'll know we can proceed.

per1234, 2019-09-11:
Closing due to lack of action. @sridharrajagopal if you ever feel like changing the repository to meet the requirements for inclusion in the Library Manager index, please comment here and I'll re-open.

closed 2019-09-11 by per1234