arduino/Arduino #9036: Disabling "check for updates" should fully disable checking for boards or libraries against the server

ucod, 2019-07-03:
Given [this](https://github.com/arduino/Arduino/issues/8988) issue, unticking "check for updates **at** startup" should disable any check against the server(s). There's absolutely no need to call upon a url for the boards or libraries since _you already berelinquish being notified of any updates at all_.

The [fix](https://github.com/arduino/Arduino/pull/9023) is kind of a joke, I mean, really, no one is going to change variants, or the compiler for a microcontroller, or a serious library, in four hours. These should actually be released with point releases such as Arduino 1.x.x etc.

matthijskooijman, 2019-07-04:
@ucod, can you be a bit more specific about what checks are still being done? From your comments, I *think* you mean that at startup, all third-party board json files might be downloaded (or maybe only with #9023 applied)? I also do not see how #8988 applies, since that, I think, is about invalid downloads when opening the board manager?

I realize that your request is to *not* do any checks against servers (which seems reasonable to me, unless those requests are not related to updates). But still, discussion about solutions becomes easier if the problem is a bit more explicit.

ucod, 2019-07-04:
Yes, simply don't do any checks at all.

matthijskooijman, 2019-07-05:
@ucod it seems you're not willing to answer my question, and provide more details?

The next step in resolving this is to have someone see whether any update checks are still happening with the checkbox unticked. Once we have a list of such checks, we can evaluate whether they should indeed be removed and discuss how to remove them (and if there any additional changes needed).

Personally, I have no time to dive into this (and I suspect this holds for the Arduino devs as well). It seems to me (though that could be my misinterpretation of the previous comment) that @ucod is not willing to provide such details, so someone is welcome to do this. If not, I'm inclined to close this issue, since it is not very actionable right now.

ucod, 2019-07-05:
@matthijskooijman Sorry. Try opening the Board or the Library manager with your network turned off and you'll get a downloading error for both of them (updates are unchecked). These are not third party. I didn't answer because I wanted to check with Wireshark.
_>why would I want to disable that_
Because I might have 2-3 different local versions of the same boards and I want to try compiling against each of them, by uninstalling and reinstalling old/new versions.

My point is that if you disable "check for updates at startup", you're actually disabling only the background checks, but you still check for updates upon opening one of those menus.

matthijskooijman, 2019-07-05:
@ucod, thanks for following up.

IMHO, checking for updates (or rather, downloading the list of available cores) when you open the board manager is sensible, even when you have disabled that checkbox (iow, I believe that checkbox rightfully does not cover this case). That still leaves the option of introducing another checkbox, specicifically for this (and perhaps similar) case(s).

However, I'm not so sure I understand your usecase. In particular, installing a particular (version of a) core, involves downloading files from the internet and installing them. To do so, you need to know what cores and versions are available. If you would *never* download the index json files, you would have nothing available to install (AFAIK the IDE does not ship any such index files).

Of course, you could skip downloading the index and use an older version if you previously downloaded that (and perhaps have an explicit "update index" button somewhere), but that would leave the possibility of outdated index files (where your old index files lists a version whose actual files are no longer available online). Though these might be cached by the IDE if you previously installed them, but that's not something you can rely on.

I'm thinking that your *actual* request here is not so much to refrain from updating, but to make it easier to switch between multiple version of a core? In that sense, I'm not sure if refraining to update the index automatically would really help, other than saving you a small bit of download (and perhaps keeping old version working even if they are removed from the index at some point)?

Perhaps the above help you better understand how things work now and offer other suggestions of things to improve your usecase?

ucod, 2019-07-05:
>you could skip downloading the index and use an older version if you previously downloaded
This and a dialog, or a button, to ask _explicitly_ for establishing a connection against the server (as in "Would like you to refresh the index from the server?"). 

I find that it simply shouldn't connect to the internet for listing items from a local file (the json) [even with a valid connection to internet available for use].

per1234, 2019-07-05:
I agree with @matthijskooijman that making it so that unchecking **File > Preferences > Check for updates during startup** would cause checking for updates when opening Library/Boards Manager to be disabled would be very unexpected behavior. Sure, we could add yet another item to preferences, but I don't think there is any need to add this additional complexity to the Arduino IDE's UI.

I can't see any problem with Boards/Library Manager trying to download the index files. The problem is that the IDE currently has a very unfriendly behavior when something goes wrong with the download of an index file. The solution is to fix the bad behavior, not to skip the download attempts.

There are two different issues I know of with the current Boards Manager system:
- Previously downloaded index file is deleted when signature verification of the fresh download of the file fails. This causes the installed boards from that index to no longer be recognized by the Arduino IDE.
- Previously downloaded index file is deleted when the fresh download of the file fails. This causes the installed boards from that index to no longer be recognized by the Arduino IDE.

We have talked about this in https://github.com/arduino/Arduino/issues/8936 and I believe https://github.com/arduino/Arduino/pull/9023 is intended to solve both of them

As I commented on that PR, the promised behavior on signature verification failure of the index file doesn't seem to be working for me currently. I also just did a test with a failed download of the index file and found that still has the same old bad behavior with the test build of https://github.com/arduino/Arduino/pull/9023 as well. However, I'm certain by the time the PR is merged it will be working.

There is some indication that at least the problem with failed download will be fixed as a matter of course when the Arduino IDE is switched to using arduino-cli (https://github.com/arduino/arduino-cli/issues/252#issuecomment-502637616). I haven't tested arduino-cli's behavior on signature verification failure yet.

ucod, 2019-07-05:
Personally, I find that _'don't connect to internet at all'_ would be the most fail-safe solution. Install the IDE once, install whatever once and then, never connect again. I could run the IDE under another user and add/remove internet connection privileges but that seems to much of a hassle.

But, ok, I won't complain further if this issue gets closed.

wavexx, 2021-01-11:
I'm hitting this issue frequently, which is causing random slowdowns of my build environment depending on network conditions.

I've resorted to using unshare(1) to completely prevent arduino to connect to the network. However this shouldn't be required:

1) if I request check.update=false no network activity should be attempted
2) still causes an exception during startup:

```
Initializing packages...
[main] WARN javax.jmdns.impl.JmmDNSImpl$NetworkChecker - Unexpected unhandled exception: java.lang.NullPointerException
```

wavexx, 2021-02-03:
In arduino 1.8.13 log4j fails to initialize without a network interface:

```
Initializing packages...
java.lang.ExceptionInInitializerError
	at org.apache.logging.log4j.core.util.WatchManager.<init>(WatchManager.java:53)
	at org.apache.logging.log4j.core.config.AbstractConfiguration.<init>(AbstractConfiguration.java:135)
	at org.apache.logging.log4j.core.config.NullConfiguration.<init>(NullConfiguration.java:32)
	at org.apache.logging.log4j.core.LoggerContext.<clinit>(LoggerContext.java:79)
	at org.apache.logging.log4j.core.selector.ClassLoaderContextSelector.createContext(ClassLoaderContextSelector.java:171)
	at org.apache.logging.log4j.core.selector.ClassLoaderContextSelector.locateContext(ClassLoaderContextSelector.java:145)
	at org.apache.logging.log4j.core.selector.ClassLoaderContextSelector.getContext(ClassLoaderContextSelector.java:70)
	at org.apache.logging.log4j.core.selector.ClassLoaderContextSelector.getContext(ClassLoaderContextSelector.java:57)
	at org.apache.logging.log4j.core.impl.Log4jContextFactory.getContext(Log4jContextFactory.java:148)
	at org.apache.logging.log4j.core.impl.Log4jContextFactory.getContext(Log4jContextFactory.java:45)
	at org.apache.logging.log4j.LogManager.getContext(LogManager.java:194)
	at org.apache.logging.log4j.LogManager.getLogger(LogManager.java:581)
	at processing.app.helpers.BoardCloudResolver.<clinit>(BoardCloudResolver.java:51)
	at cc.arduino.packages.discoverers.serial.SerialDiscovery.<init>(SerialDiscovery.java:49)
	at cc.arduino.packages.DiscoveryManager.<init>(DiscoveryManager.java:52)
	at processing.app.BaseNoGui.initPackages(BaseNoGui.java:508)
	at processing.app.Base.<init>(Base.java:278)
	at processing.app.Base.main(Base.java:150)
Caused by: java.lang.NullPointerException
	at org.apache.logging.log4j.core.util.NetUtils.getMacAddress(NetUtils.java:97)
	at org.apache.logging.log4j.core.util.UuidUtil.<clinit>(UuidUtil.java:67)
	... 18 more
```