arduino/Arduino #9037: Acessibility: Refactor UI to provide alternative to dropdown menus

sbrl, 2019-07-05:
I'm using the Arduino IDE with an external editor, and I find myself constantly going through the dropdown menus at the top to do things like change the board, and selecting the port.

Sometimes I have to dig through multiple menu levels to find the option I'm after - selecting a different board is particularly painful as I've got to wait for the dropdown menu to scroll down _every single time_ (and because the IDE doesn't integrate application menus with dbus, I can't even search the menus with my OS).

I suggest refactoring the UI to add dedicated buttons to select a board (which would bring up a separate dialogue box with a searchable list of boards), and the port to use.

per1234, 2019-07-05:
Since it's a feature request for the Arduino IDE, this indeed the correct place to submit the issue. Thanks!

There is a little known feature added since Arduino IDE 1.8.6 that might make using the current menu system a bit easier. If you hold the 'a' key while scrolling a menu it accelerates the scroll speed (https://github.com/arduino/Arduino/blob/master/build/shared/revisions.txt#L68).

There is an open pull request that will add a list of recently used boards at the top of the **Tools > Board** menu. The recently used boards are each assigned a keyboard shortcut (https://github.com/arduino/Arduino/pull/8607). The data is not saved between sessions so you will need to go through the normal board selection process after every time you restart the Arduino IDE to populate the recently used boards section. This feature is already integrated in the beta version of the Arduino IDE if you want to try it out:
https://www.arduino.cc/en/Main/Software#beta

sbrl, 2019-07-05:
Phew! Glad I got the right place :P

Ah, interesting! I didn't know about that. It goes help, but when you use it it jumps really quickly and you lose your place.

Yeah, a recently used boards section would help, but considering the number of boards that are out there these days I think the best solution is to have a searchable list.