arduino/Arduino #9039: Manage Library Manager requests elsewhere

NicoHood, 2019-07-05:
I do track this repository to get notified about interesting things. But those library manager requests spam my github notifications a lot. Could you please setup a proper formular or another github repository for those requests?

per1234, 2019-07-05:
The "another GitHub repository" request is a duplicate of https://github.com/arduino/Arduino/issues/6332

What do you mean by "a proper formular"?

NicoHood, 2019-07-13:
A formular on the website, instead of "abusing" githubs issue tracker. No matter what, it should be moved out of this bug tracker. I will close this ticket, sorry for the duplicate.

per1234, 2019-07-13:
I'm not familiar with the term "formular", but I think it would be an excellent idea to have a form on the web page to enter Library Manager addition requests. Ideally, the entire process would be automated, including feedback if the library was rejected due to not meeting all the requirements for addition to the index. With >2300 libraries added so far and no sign of it slowing (my last automated scan of Arduino libraries on GitHub discovered over 10000, so there are plenty more, and new ones coming every day), surely the time to develop an automated system would be well invested.

Although the idea of a library submission form might have come up in passing before (example: https://github.com/arduino/Arduino/issues/2854#issuecomment-87966744), I don't find a dedicated issue about this feature request, so I don't think that part of your issue was a duplicate. For that reason, I'm going to re-open this. I hope you won't mind.

per1234, 2021-06-07:
We didn't end up going with the formular approach, but I think [the new Library Manager submission system](https://forum.arduino.cc/t/new-library-manager-submission-update-system/871710) is in the spirit of this request.

The pull request approach is a little bit more complex than a glossy form on a website, but I think that is offset by the increased transparency this approach provides. Most of those who submit libraries will be familiar with the pull request process and so can observe every step of the library's journey through the Library Manager, from submission to installation by the user.

closed 2021-06-07 by per1234