arduino/Arduino #9042: Please add ArduinoEducation Library

jsmith2021Brandeis, 2019-07-08:
Please Add My Arduino Library to the library manager:
https://github.com/jsmith2021Brandeis/ArduinoEducation

This project includes libraries and example files designed to introduce programming Arduinos to new students. The classes demonstrate arrays, a distance sensor, and a display.

per1234, 2019-07-08:
Hi. Arduino recently started a new policy that 3rd party libraries with a library.properties `name` value starting with the word "Arduino" (or any capitalization thereof) will no longer be accepted for inclusion in the Arduino Library manager index. The reason is that Arduino wants to reserve these names for their own official libraries.

If you want your library to be added to the Library Manager index, you will need to change the `name` value in library properties to one that is compliant with the Arduino Library Specification:
https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5:-Library-specification#libraryproperties-file-format

Then update the `version` value in library.properties and create a new release/tag, or else redo the existing release/tag.

Note that this restriction only applies to the `name` value. You are not required to rename the repository or header files.

Once you have done this, please comment here so we'll know we can proceed.

More information:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

jsmith2021Brandeis, 2019-07-08:
Ok, I updated the name with a non Arduino name
And made a new release 1.0.1

cmaglie, 2019-07-17:
done

closed 2019-07-17 by cmaglie