arduino/Arduino #9051: Please add the 8 Digit VFD library to the Arduino Library Manager

earthtown, 2019-07-09:
Please add the [8 Digit VFD library](https://github.com/earthtown/8_digit_vfd/releases/tag/v1.0.0) to the Arduino Manager when you get a chance!
Thanks

per1234, 2019-07-09:
Hi. In order to be added to the Arduino Library Manager index, your library must be in the root of the repository. Your library is currently under the `VFD_8digit` subfolder, which is not allowed. Please move your library up a folder level and then either redo the existing release, or create a new release (after updating the `version` value in library.properties).

earthtown, 2019-07-09:
I have corrected the problem and created a new release. Do I need to create
a new issue?

Thanks for helping me with my first submission to the Arduino Library
Manager!
-Brandon

per1234, 2019-07-09:
Thanks. There's no need to create a new issue.

cmaglie, 2019-07-17:
done

closed 2019-07-17 by cmaglie