arduino/Arduino #9058: [Library Manager] please add Atlas EC_OEM library.

andhieSetyabudi, 2019-07-12:
https://github.com/andhieSetyabudi/atlas_OEM

per1234, 2019-07-12:
Hi. There are a few problems that will need to be resolved before your library meets the requirements for addition to the Arduino Library Manager index:
- Your `version` value in library.properties is non-compliant. It cannot contain leading zeros so `1.0.01` is invalid. `1.0.1` would be the correct form.
- Leading space on the `name` field. According to the [Arduino Library specification](https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5:-Library-specification#libraryproperties-file-format), this is not allowed. I actually am not certain this requirement is real, since I've never noticed a request get blocked over it and I'd be amazed if there were no other libraries with leading space in the >2000 that have been added to the index. However, since you're going to be editing library.properties anyway, it's safest to fix this too. Besides, the leading space there serves no good purpose so it needs to go anyway.
- Your repository doesn't contain a [release](https://help.github.com/articles/creating-releases/) or [tag](https://git-scm.com/docs/git-tag). After fixing the above problems, please create a release or tag that matches the `version` value in library.properties.

After doing so, please comment here to let us know we can proceed with adding your library to the Library Manager index.

For more information see:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

andhieSetyabudi, 2019-07-12:
thanks, i just fix it

cmaglie, 2019-07-17:
done

closed 2019-07-17 by cmaglie