arduino/Arduino #9060: [Library Manager] ESP32 Rest Client into Library Manager

eduardomarcos, 2019-07-12:
https://github.com/eduardomarcos/arduino-esp32-restclient

Regards!

per1234, 2019-07-12:
Hi. Arduino recently started a new policy that 3rd party libraries with a library.properties `name` value starting with the word "Arduino" (or any capitalization thereof) will no longer be accepted for inclusion in the Arduino Library manager index. The reason is that Arduino wants to reserve these names for their own official libraries.

If you want your library to be added to the Library Manager index, you will need to change the `name` value in library properties to one that is compliant with the Arduino Library Specification:
https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5:-Library-specification#libraryproperties-file-format

Then update the `version` value in library.properties and create a new release/tag, or else redo the existing release/tag.

Note that this restriction only applies to the `name` value. You are not required to rename the repository or header files.

Once you have done this, please comment here so we'll know we can proceed.

More information:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

eduardomarcos, 2019-07-13:
Hi!

Renamed with the same tag :) Hope is it good now.

Thank you in advance!

cmaglie, 2019-07-17:
done

closed 2019-07-17 by cmaglie