arduino/Arduino #9061: Add "Senses_wifi" library to Library manager Arduino IDE

Isaranu, 2019-07-13:
We are developer team from Senses Innovation which provided an "Senses IoT platform" by sending data from IoT device.
It's compatible use with ESP8266 Wifi module.

github url : https://github.com/Isaranu/Senses_wifi

Thank you very much.
Senses team.

per1234, 2019-07-13:
You need to create a [release](https://help.github.com/articles/creating-releases/) or [tag](https://git-scm.com/docs/git-tag) that matches the `version` value in your library.properties file.

After doing so, please comment here to let us know we can proceed with adding your library to the Library Manager index.

For more information see:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

Isaranu, 2019-07-14:
> You need to create a [release](https://help.github.com/articles/creating-releases/) or [tag](https://git-scm.com/docs/git-tag) that matches the `version` value in your library.properties file. [...]

Hi,
I'm so sorry about forget this.
By the way, I do a version release now.
Please check it again.

Best Regards,
Senses teams

cmaglie, 2019-07-17:
done

closed 2019-07-17 by cmaglie