arduino/Arduino #9062: jSSC-2.8_x86.dll is installed in a new User folder in Windows

gbhug5a, 2019-07-13:
jSSC-2.8_x86.dll is installed in a new ".jssc" folder under User\Owner\  even though the dll already appears in Arduino\lib\.  Is this a permissions issue or something similar?  If not, it would help keep the User folders clean if the dll could be run from the \lib folder without creating a new user-level folder for it that just clutters up the Windows Explorer tree.

v1.8.8 Portable, Windows 7/64.

per1234, 2019-07-13:
This issue had been fixed by https://github.com/arduino/Arduino/commit/229c32ccad40f3e7b9bce937bef15808e3e0611c, but then there was a regression in Arduino IDE 1.8.7, which I suspect was caused by https://github.com/arduino/Arduino/commit/6080907ba36e1cfcf0c90afe1f61e9546904f78d

Related: https://github.com/arduino/Arduino/issues/6626

facchinm, 2019-07-15:
It's very strange since jssc `-arduino2` was just a repackaging of `-arduino1` with aarch64 lib added and no extra rule has been added/removed from ant script. I'll investigate a bit more and report you back.

facchinm, 2019-07-16:
@per1234 you were totally right, I repackaged plain jssc at the time of adding aarch64 lib, not `arduino1` . Fixed by https://github.com/arduino/Arduino/commit/fb9ce0ade323f6f36793f29d615db1272d184ba4

closed 2019-07-16 by facchinm

justintconroy, 2020-05-01:
If I have installed the latest release of the Arduino IDE, can I safely delete this from my home directory?

commit fb9ce0ad (2019-07-16): Update jssc to arduino4

Repackage jssc-2.8.0-arduino1 and add aarch64 lib instead than plain jssc

Fixes #9062