arduino/Arduino #9065: New Arduino Library for the MAX5417/18/19 Digital Potentiometers

hoilett, 2019-07-14:
Hi,

The library is located here: https://github.com/LinnesLab/MAX541X

I created a library for the MAX5417/18/19 digital potentiometers from Maxim Integrated. These are 256-tap, 50k, 100k, or 200k digital potentiometers that can be used to create programmable gain amplifiers (PGAs) and for general device control. I personally use them as PGAs in my smartwatch project: https://ieeexplore.ieee.org/abstract/document/8513356

The library contains all the necessary functions for users to get started with the device and provides relevant examples as well. I have also included the EAGLE files for a breakout board.

The library is located here: https://github.com/LinnesLab/MAX541X

per1234, 2019-07-14:
Hi. I notice the `paragraph` field is missing from your library.properties file. I believe this will block inclusion in the Arduino Library Manager index. Please add a `paragraph` field, then either create a new release or tag or else redo the existing tag.

Once that's done, please comment here so we'll know we can proceed.

More information:
https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5:-Library-specification#libraryproperties-file-format

hoilett, 2019-07-14:
Hi there,
I added the paragraph field. I've also added a new release v1.0.2 (I created a v.1.0.1 tag prematurely) and updated the version number in the library.properties file to v1.0.2 accordingly.

cmaglie, 2019-07-17:
done

closed 2019-07-17 by cmaglie

hoilett, 2019-07-17:
Schweet! Thanks.