arduino/Arduino #9066: [Library Manager] Please add SparkFun Temperature Sensor - TMP117 - Arduino Library

madichodikov, 2019-07-15:
Hello, 

Please add our TMP117 library to the library manager. Here is the link: 
https://github.com/sparkfunX/SparkFun_TMP117_Arduino_Library

Many thanks!

per1234, 2019-07-15:
Hi. There are a few issues that will need to be resolved before the library meets the requirements for addition to the Arduino Library Manager index:
- The `name` value in library.properties contains disallowed characters: `(` and `)`. According to the [Arduino Library specification](https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5:-Library-specification#libraryproperties-file-format):
  > the name of the library. Library names must contain only basic letters (A-Z or a-z) and numbers (0-9), spaces ( ), underscores (_), dots (.) and dashes (-). It cannot start or end with a space, and also it cannot start with a number.
- The `version` value in library.properties is invalid. According to the Arduino Library specification:
  > Version should be [semver ](http://semver.org/)compliant. 1.2.0 is correct; 1.2 is accepted; r5, 003, 1.1c are invalid
- Once you have corrected the `name` and `version` values, you must create a [release](https://help.github.com/articles/creating-releases/) or [tag](https://git-scm.com/docs/git-tag) that matches the `version` value in your library.properties file.

After completing the above, please comment here to let us know we can proceed with adding your library to the Library Manager index.

For more information see:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

madichodikov, 2019-07-16:
Thanks, it should be fixed now.

madichodikov, 2019-07-16:
Hi, please note that the repo location has changed
https://github.com/sparkfun/SparkFun_TMP117_Arduino_Library

cmaglie, 2019-07-17:
done

closed 2019-07-17 by cmaglie