arduino/Arduino #9073: [Library Manage]

dushyantahuja, 2019-07-17:
Please add the following library:

https://github.com/dushyantahuja/TFL-Status

Thanks

per1234, 2019-07-18:
Hi. I believe the lack of a `maintainer` field in your library.properties file will block admission of your library into the Library Manager index.

I submitted a pull request to resolve this issue: https://github.com/dushyantahuja/TFL-Status/pull/1

Please add this field and then either create a new tag/release or else redo the existing release.

Once that's done, please comment here so we'll know we can proceed.

More information:
https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5:-Library-specification#libraryproperties-file-format

dushyantahuja, 2019-07-18:
Thanks a lot. I have updated the tag

dushyantahuja, 2019-07-23:
Can you please add now.

matthijskooijman, 2019-07-23:
Library manager updates are done periodically in bigger batches, so it might be a few more days or weeks until this is done.

dushyantahuja, 2019-07-23:
Thanks a lot

cmaglie, 2019-08-02:
done

closed 2019-08-02 by cmaglie