arduino/Arduino #9075: Add "Senses_wifi_esp32" library to Library manager Arduino IDE

Isaranu, 2019-07-18:
We are developer team from Senses Innovation which provided an "Senses IoT platform" by sending data from IoT device.
It's compatible use with ESP32 Wifi module.

github url : https://github.com/Isaranu/Senses_wifi_esp32

Thank you very much.
Senses team.

per1234, 2019-07-18:
Hi. I believe the lack of a `paragraph` field in your library.properties file will block admission of your library into the Library Manager index.

Please add this field and then either create a new tag/release or else redo the existing release.

Once that's done, please comment here so we'll know we can proceed.

More information:
https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5:-Library-specification#libraryproperties-file-format

Isaranu, 2019-07-19:
> Hi. I believe the lack of a `paragraph` field in your library.properties file will block admission of your library into the Library Manager index.
>
> Please add this field and then either create a new [...]

Hi,
I'm so sorry about missing article in library.properties.
There was added information.
And I released version to v0.1.1
Please check it again

Best Regards,
Senses team

Isaranu, 2019-07-25:
I added more examples code.
So, I released it to v.0.1.2 (Already released tag and revised library.properties)

Please add this lib to Arduino Lib manager.

Thank you :)

Isaranu, 2019-07-26:
I revised some code.
So, I changed version to 0.2.0
Done both of library.properties and tag version released

cmaglie, 2019-08-02:
done

closed 2019-08-02 by cmaglie