arduino/Arduino #9077: Want to correct library name of WiredController_asukiaaa

asukiaaa, 2019-07-18:
I missed spell of `WiredController_asukiaaa` as `WiredContriller_asukiaaa`.
I correct the spell and release new version but library system does not recognize it.

https://github.com/asukiaaa/WiredController_asukiaaa/releases/tag/1.0.1
https://github.com/asukiaaa/WiredController_asukiaaa/commit/05dfcf418cf54d70e57d96a555d7f1ec2104925e

Can I rename library with connect spell?

per1234, 2019-07-18:
As you discovered, the Library Manager locks your library into the `name` value on the tag imported to the index when your library is first added so if you change the `name` value in subsequent tags it doesn't change the library name in Library Manager.

The library name can be manually changed on request though. cmaglie will do this on the next run through the Library Manager requests.

asukiaaa, 2019-07-18:
Thank you, I'm glad to know that.
I wait next running about Library Manager.

cmaglie, 2019-08-02:
done

closed 2019-08-02 by cmaglie

asukiaaa, 2019-08-03:
Thank you for changing.