arduino/Arduino #9081: Please Add "TEE UC20 Shield" to Library manager Arduino IDE.

ThaiEasyElec, 2019-07-19:
Please Add "TEE UC20 Shield" to Library manager Arduino IDE.

github url : https://github.com/ThaiEasyElec/TEE_UC20_Shield

Thank you very much.
support@thaieasyelec.com

per1234, 2019-07-19:
In order to be added to the Library Manager index your library must contain a library.properties file. Although your repository does contain a library.properties file now, it didn't at the time of your 0.9 release. Please either create a new [release](https://help.github.com/articles/creating-releases/) or [tag](https://git-scm.com/docs/git-tag) or else redo the existing tag. Make sure that the release version matches the `version` value in your library.properties at the time of the release.

Once you have done this please comment here so we know we can proceed.

ThaiEasyElec, 2019-07-19:
Thank you for suggestion. now i fix it. 
please check again.

cmaglie, 2019-08-02:
done

closed 2019-08-02 by cmaglie