arduino/Arduino #9082: Upload error problems that not happens with older Arduino versions

avandalen, 2019-07-19:
See the article here:
https://www.avdweb.nl/arduino/misc/troubleshooting
Some years ago, I used an Arduino Nano board. Now I wanted to use this board again. But suddenly, error messages occur when uploading a sketch. That is strange, it has always worked before! 
The same happens with an Arduino Pro-Mini board.

I have found out what the actual problem is: New Arduino software versions such as v1.8.9 are not compatible with older versions

After I had removed the Arduino version 1.8.9 from 2019 and installed an old version 1.8.5 from 2017, the upload problem was solved. 

For resolving upload errors, it would be useful if the Arduino IDE could automatically switch to the correct communication protocol for the bootloader.

per1234, 2019-07-19:
For the problem you're having with using the old Nano with the new IDE, it's most likely that you didn't select **Tools > Processor > ATmega328P (Old Bootloader)**. Give that a try and let us know if it solves your problem. This information is documented on the Nano's Getting Started page:
https://www.arduino.cc/en/Guide/ArduinoNano#toc4

I don't know what the problem is with your Pro Mini, since Arduino has never changed the bootloader. In order to investigate this problem, we'd need you to provide some information. Please do this:
1. File > Preferences
1. Uncheck the box next to **Show verbose output during: > compilation**
1. Check the box next to **Show verbose output during: > upload**
1. Click the "OK" button.
1. Sketch > Upload
1. After the upload fails you'll see a button on the right side of the orange bar "Copy error messages". Click that button.
1. [Paste](https://lifehacker.com/how-to-copy-cut-and-paste-for-beginners-5801525) the error messages in a reply here [using code fencing](https://guides.github.com/features/mastering-markdown/).

avandalen, 2019-07-25:
I per1234

I have less time, maybe i can send you my arduino nano, (which you can keep)
than you can use this board to test new arduino versions?

kind regards

Albert

From: per1234 
Sent: Friday, July 19, 2019 12:01 PM
To: arduino/Arduino 
Cc: Albert van Dalen ; Author 
Subject: Re: [arduino/Arduino] Upload error problems that not happens with older Arduino versions (#9082)

For the problem you're having with using the old Nano with the new IDE, it's most likely that you didn't select Tools > Processor > ATmega328P (Old Bootloader). Give that a try and let us know if it solves your problem. This information is documented on the Nano's Getting Started page:
https://www.arduino.cc/en/Guide/ArduinoNano#toc4

I don't know what the problem is with your Pro Mini, since Arduino has never changed the bootloader. In order to investigate this problem, we'd need you to provide some information. Please do this:

  1.. File > Preferences 
  2.. Uncheck the box next to Show verbose output during: > compilation 
  3.. Check the box next to Show verbose output during: > upload 
  4.. Click the "OK" button. 
  5.. Sketch > Upload 
  6.. After the upload fails you'll see a button on the right side of the orange bar "Copy error messages". Click that button. 
  7.. Paste the error messages in a reply here using code fencing.

per1234, 2019-07-25:
It's true that having the hardware on hand makes troubleshooting much easier than trying to direct the troubleshooting procedure remotely. However, from your website it appears you are located in the Netherlands, while I'm in the US, so the shipping cost to send me the Nano might be prohibitive.

avandalen, 2019-07-26:
I will sent it you for free and free shipping

From: per1234 
Sent: Friday, July 26, 2019 12:11 AM
To: arduino/Arduino 
Cc: Albert van Dalen ; Author 
Subject: Re: [arduino/Arduino] Upload error problems that not happens with older Arduino versions (#9082)

It's true that having the hardware on hand makes troubleshooting much easier than trying to direct the troubleshooting procedure remotely. However, from your website it appears you are located in the Netherlands, while I'm in the US, so the shipping cost to send me the Nano might be prohibitive.

per1234, 2019-07-26:
I sent and email to the address on the contact page of your website. I'll update this issue report with the findings of my investigation.

per1234, 2019-08-14:
The upload to the Nano works fine with **Tools > Processor > ATmega328P (Old Bootloader)** selected.

closed 2019-08-14 by per1234