arduino/Arduino #9092: [Library Manager] Adafruit fork of SdFat

ladyada, 2019-07-23:
please add to library manager - thanx!
https://github.com/adafruit/SdFat/blob/master/library.properties

per1234, 2019-07-23:
Hi. The characters `(` and `)` used in the library.properties `name` field are not allowed.

From the Arduino Library Specification:
https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5:-Library-specification#libraryproperties-file-format
>**name** - the name of the library. Library names must contain only basic letters (A-Z or a-z) and numbers (0-9), spaces ( ), underscores (_), dots (.) and dashes (-). It cannot start or end with a space, and also it cannot start with a number.

Please change the `name` value to something that doesn't use any disallowed characters and then either create a new release or else redo the existing release. Once that is done, please comment here so we'll know we can proceed.

ladyada, 2019-07-27:
ahh sorry! fixed now and bumped to 1.2.1 :)

cmaglie, 2019-08-02:
done

closed 2019-08-02 by cmaglie