arduino/Arduino #9095: [Library Manager] Please Add Library : Library for GoGoBoard

thapakorn613, 2019-07-23:
Hello , Please add the following library to the Library Manager index:
.
https://github.com/thapakorn613/GoGoBoard-Library
.
Thanks

per1234, 2019-07-23:
You need to create a [release](https://help.github.com/articles/creating-releases/) or [tag](https://git-scm.com/docs/git-tag) that matches the `version` value in your library.properties file.

After doing so, please comment here to let us know we can proceed with adding your library to the Library Manager index.

For more information see:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

thapakorn613, 2019-07-24:
i made it , Please add my library again 
.
Thanks

cmaglie, 2019-08-02:
done

closed 2019-08-02 by cmaglie

njh, 2020-10-31:
Hi @thapakorn613, it looks like you have deleted this library from Github.
Would you mind letting me know why?