arduino/Arduino #9097: Arduino IDE 1.8.9 Yun Shield

silverar, 2019-07-24:
Hello there

I can't upload sketch over wifi from the IDE 1.8.9 while in IDE 1.8.5 works with no problem.
In 1.8.9 i get this issue when i click upload : 

processing.app.debug.RunnerException
	at cc.arduino.packages.uploaders.SSHUploader.uploadUsingPreferences(SSHUploader.java:147)
	at cc.arduino.UploaderUtils.upload(UploaderUtils.java:77)
	at processing.app.SketchController.upload(SketchController.java:732)
	at processing.app.SketchController.exportApplet(SketchController.java:703)
	at processing.app.Editor$UploadHandler.run(Editor.java:2070)
	at java.lang.Thread.run(Thread.java:748)
Caused by: java.lang.NullPointerException
	at cc.arduino.packages.ssh.SSHConfigFileSetup.setup(SSHConfigFileSetup.java:49)
	at cc.arduino.packages.uploaders.SSHUploader.uploadUsingPreferences(SSHUploader.java:112)
	... 5 more

silverar, 2019-07-24:
In both version i input the password but in 1.8.9 will get error.

facchinm, 2019-07-30:
Hi @silverar ,
which board is under the Yun shield? And which core version are you using? Thanks!

facchinm, 2019-07-30:
I found the commit originating the bug https://github.com/arduino/Arduino/commit/cfd3cf2b276b5d607d055112c71ef52408c2058e , it is due to the Yun Shiled not being a proper board but just a shield. The fix is ongoing :wink:

commit adabb083 (2019-07-30): NetworkUpload. use ipAddress to start jSch session

Fixes #9097
closed 2019-07-30 by facchinm