arduino/Arduino #9102: Serial Monitor Feature Request

online-data-archive, 2019-07-27:
The Serial Monitor as it stands just display Serial Data and accepts data viar a command line.

Can the Serial Monitor have the ability to Respond to Terminal Escape codes
Ie Font Change Cursor position, Draw lines, change font colour.

I realise that a external Terminal could be run but once the External Serial Terminal has taken
control of the port, the uploading and download of code cannot be accomplished,
Whereas the Serial Monitor looks to have controls to Disconnect automatically while code
is uploaded.

Hiddenvision, 2019-08-07:
It would be nice to have a Hex/Dec/Char toggle also

per1234, 2020-10-06:
Closing as duplicate of https://github.com/arduino/Arduino/issues/346 and https://github.com/arduino/Arduino/issues/3654.

closed 2020-10-06 by per1234