arduino/Arduino #9107: [Library Manager] Add library to Arduino Library Manager

brigosx, 2019-07-30:
Hello, please add the following library to Arduino Library Manager:

https://github.com/brigosx/SevenSeg4D.git

Thanks

cmaglie, 2019-08-02:
You need to create a [release](https://help.github.com/articles/creating-releases/) or [tag](https://git-scm.com/docs/git-tag) that matches the `version` value in your library.properties file.

After doing so, please comment here to let us know we can proceed with adding your library to the Library Manager index.

For more information see:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

brigosx, 2019-08-03:
Dear cmaglie,

Thank you very much for your guidance. I've just prepared (hopefully) a release for the version of my library by following a GIT guide I found.

Thanks again for everything.

cmaglie, 2019-08-21:
done, sorry for the delay!

closed 2019-08-21 by cmaglie