arduino/Arduino #9112: Serial interface locks up on Nano Every

technoblogy, 2019-07-31:
On a Nano Every, if you try and transfer more than about 128 characters from the Serial Monitor to the board via the serial interface it locks up, and the only solution is to unplug the USB cable.

The following program demonstrates the problem:

    void setup() {
      Serial.begin(9600);
      while (!Serial);
      Serial.println("Ready");
    }

    int count = 0;

    void loop() {
      if (count % 64 == 0) {
        Serial.println();
        Serial.print(count);
        Serial.write(' ');
      }
      while (!Serial.available());
      Serial.write((int)Serial.read());
      count++;
    }

Open the **Serial Monitor**, copy a line of text of 128 characters, paste it into the input field, and click **Send**.

I'm running Arduino IDE 1.8.9 on a Mac, and the Arduino megaAVR Boards file version 1.8.3.

It works fine on a Uno WiFi Rev 2.

facchinm, 2019-08-02:
Hi @technoblogy , thanks for reporting. I investigated the problem and indeed it lies in the USB stack for the D11 (muxto).
I patched the binaries here (https://github.com/arduino/ArduinoCore-megaavr/commit/ee0a9bdc4ff6e30fc2fe7f53025663c737a5e37d) but I would need your help to validate them.

Could you follow the guide at https://www.arduino.cc/en/Guide/NANOEvery (Firmware for SAMD11D14A chapter), trigger the D11 bootloader, extract the attached zip and execute this command?
```./bossac --port=$portname -U true -i -e -w -v MuxTO.bin -R```
[update_muxto.zip](https://github.com/arduino/Arduino/files/3461449/update_muxto.zip)
Thanks a lot

technoblogy, 2019-08-03:
Thanks for the reply - I'll try upgrading my firmware.

technoblogy, 2019-08-16:
@facchinm I'm struggling with this. Do I have to install the MattairTech core? Do I need to solder a link between the D11 SWDIO and GND pins? Where should I give the command:

`./bossac --port=$portname -U true -i -e -w -v MuxTO.bin -R`

When I do it in the Mac Terminal I just get:

`./bossac: No such file or directory`

technoblogy, 2019-08-17:
@facchinm I'm persevering, and tried downloading and installing Bossa 1.9.1 for Mac from here:

`https://github.com/shumatech/BOSSA/releases`

However on running it I get this error:

[image]

facchinm, 2019-08-19:
Hi @technoblogy ,
the first command you used was fine (you need the `bossac` binary contained in the zip).
Probably OSX blocked the file from being executable, so my advice is:
* run `chmod +x bossac` inside the extracted folder
* run `./bossac` and see if it starts at least
* no need to solder the link, the short circuit should exist just when powering the board (I normally use tweezers to create the short and then connect the type A side of the USB cable to the PC, with the cable on the nano side already inserted)
* `$portname` should be replaced by the actual port name, that you can discover from the IDE serial monitor

technoblogy, 2019-08-19:
Thanks, I'll try it later today.

technoblogy, 2019-08-19:
Still not working. I plugged in the board with the SWDIO pin connected to ground. The instructions say:

>  The board will appear as "MuxTo bootloader"

Where will that appear?

I'm getting:

    david$ ./bossac --port=/dev/cu.usbmodem1411 -U true -i -e -w -v MuxTO.bin -R
    No device found on /dev/cu.usbmodem1411

technoblogy, 2019-08-19:
OK, I've worked it out. The Serial Monitor says /dev/cu.usbmodem1411, but if I do:

`ls /dev/tty.*`

I get:

`/dev/tty.Bluetooth-Incoming-Port	/dev/tty.usbmodem1411`

So I tried:

`./bossac --port=/dev/tty.usbmodem1411 -U true -i -e -w -v MuxTO.bin -R`

and that worked.

technoblogy, 2019-08-19:
I confirm that the muxto bootloader 1.0.6 fixes the serial problem I reported - many thanks!

One final question. I was thinking of specifying the Nano Every board to users for an application I've developed, but for my application this problem makes the board unusable, and I don't think they will all be able to do this upgrade.

How many boards have shipped with the pre-1.0.6 bootloader? Will it be corrected in future boards, and will there be a way of telling which version you've got?

Thanks, David

facchinm, 2019-08-20:
Thanks for getting back!

The next batch will be flashed with 1.0.6 but there's very little chance to be able to reflash the boards currently in the market (except the ones we still have "in house"). 

About reflashing, the procedure you followed is the most complicated (but doesn't need any additional hardware). Any 10$ SWD programmer can reflash the D11 using the same pins on the bottom and is much faster (if you need to do it more than once).

Finding the MuxTO version flashed on a board is not easy at all, we had to strip a lot of information to fit the firmware in 12KB, but we can trace back the fw that was originally flashed using the board's serial number.

technoblogy, 2019-08-20:
> we can trace back the fw that was originally flashed using the board's serial number.

I think that would be useful, so we can say "if your serial number is later than xxx you don't need to do this".

facchinm, 2019-08-20:
It's a bit more complicated since serial numbers are not consecutive, so we need to query the boards DB to get that information :expressionless:

technoblogy, 2019-08-20:
Here's a summary of the procedure that worked for me on a Mac, in case it's useful for anyone else:

- Download MuxTO 1.0.6 from [update_muxto.zip](https://github.com/arduino/Arduino/files/3461449/update_muxto.zip).

- Drag the downloaded folder to the desktop.

- Open Mac Terminal.

- Change to the update_muxto folder with the command (insert your user name as appropriate):

`cd /Users/david/Desktop/update_muxto`

- While connecting the SWDIO pin to GND connect the board to your computer's USB port. You can then remove the link:

[image]

- In the Terminal find the USB port by giving the command:

`ls /dev/tty.*`

- This should print a list of ports; something like:

`/dev/tty.Bluetooth-Incoming-Port	/dev/tty.usbmodem1411`

The one you want is the usbmodem one.

- Give the command:

`./bossac --port=/dev/tty.usbmodem1411 -U true -i -e -w -v MuxTO.bin -R`

substituting the port printed in the previous command.

If all is well you should see the upgrade information printed by the Bossac installer. When that's finished you can unplug the board, and then plug it back in to use it with the new bootloader.

technoblogy, 2019-08-28:
I now have a different problem; detecting whether the Serial is ready.

With this simple test:

````
void setup() {
  Serial.begin(9600);
}

void loop() {
  Serial.println("Hello");
  for(;;);
}
````
I get nothing printed unless I insert a delay:
````
void setup() {
  Serial.begin(9600);
  delay(5000);
}

void loop() {
  Serial.println("Hello");
  for(;;);
}
````
The usual test for whether the Serial is ready is:
 ````
while (!Serial);
````
However, on Nano Every it appears that **Serial** returns true even when the serial isn't ready, because if I do:
````
void setup() {
  Serial.begin(9600);
  int start = millis();
  while ((millis() - start) < 5000) { if (Serial) break; }
}

void loop() {
  Serial.println("Hello");
  for(;;);
}
````
I again get nothing printed.

technoblogy, 2019-08-28:
Although MuxTO 1.0.6 is a big improvement with the version that was originally installed on my Nano Every, unfortunately there is still corruption occurring in the serial interface. You can demonstrate this as follows:

- Load and run the original demonstration program:

````
void setup() {
  Serial.begin(9600);
  while (!Serial);
  Serial.println("Ready");
}

int count = 0;

void loop() {
  if (count % 64 == 0) {
    Serial.println();
    Serial.print(count);
    Serial.write(' ');
  }
  while (!Serial.available());
  Serial.write((int)Serial.read());
  count++;
}
````

- Select and copy the following text:

````
 !"#$%&'()*+,-./0123456789:;<=>?@ABCDEFGHIJKLMNOPQRSTUVWXYZ[\]^_`abcdefghijklmnopqrstuvwxyz{|}~
 !"#$%&'()*+,-./0123456789:;<=>?@ABCDEFGHIJKLMNOPQRSTUVWXYZ[\]^_`abcdefghijklmnopqrstuvwxyz{|}~
 !"#$%&'()*+,-./0123456789:;<=>?@ABCDEFGHIJKLMNOPQRSTUVWXYZ[\]^_`abcdefghijklmnopqrstuvwxyz{|}~
 !"#$%&'()*+,-./0123456789:;<=>?@ABCDEFGHIJKLMNOPQRSTUVWXYZ[\]^_`abcdefghijklmnopqrstuvwxyz{|}~
 !"#$%&'()*+,-./0123456789:;<=>?@ABCDEFGHIJKLMNOPQRSTUVWXYZ[\]^_`abcdefghijklmnopqrstuvwxyz{|}~
 !"#$%&'()*+,-./0123456789:;<=>?@ABCDEFGHIJKLMNOPQRSTUVWXYZ[\]^_`abcdefghijklmnopqrstuvwxyz{|}~
 !"#$%&'()*+,-./0123456789:;<=>?@ABCDEFGHIJKLMNOPQRSTUVWXYZ[\]^_`abcdefghijklmnopqrstuvwxyz{|}~
 !"#$%&'()*+,-./0123456789:;<=>?@ABCDEFGHIJKLMNOPQRSTUVWXYZ[\]^_`abcdefghijklmnopqrstuvwxyz{|}~
 !"#$%&'()*+,-./0123456789:;<=>?@ABCDEFGHIJKLMNOPQRSTUVWXYZ[\]^_`abcdefghijklmnopqrstuvwxyz{|}~
 !"#$%&'()*+,-./0123456789:;<=>?@ABCDEFGHIJKLMNOPQRSTUVWXYZ[\]^_`abcdefghijklmnopqrstuvwxyz{|}~
 !"#$%&'()*+,-./0123456789:;<=>?@ABCDEFGHIJKLMNOPQRSTUVWXYZ[\]^_`abcdefghijklmnopqrstuvwxyz{|}~
````

- Open the Arduino IDE Serial Monitor. Make sure it's set to **Newline**, **9600 baud**.

- Paste the text into the field at the top of the Serial Sonitor and click **Send**.

The output looks like this; the red circle shows the dropped characters.

[image]

hiduino, 2019-08-29:
Is there any Windows driver for the MuxTo mode?  Or do I need to find a Mac to update the firmware?

facchinm, 2019-09-09:
@technoblogy about the `!Serial` issue there's little we can do with the current [schematics](https://content.arduino.cc/assets/NANOEveryV3.0_sch.pdf) since the RST pin is not connected. A possible workaround is invoking the UPDI reset via fw on port open but the space to implement it is very tight.

Same space issue for holes in transmission, samd11 RAM is not enough to have big buffers, so even if all the data is flushed it still takes the "9600bps-specific" amount of time to deliver it, and the serial starts dropping characters.

@hiduino you can use the exact same package for windows, simply replace `bossac` binary with `bossac.exe` from [here](https://www.mattairtech.com/software/arduino/bossac-1.7.0-mattairtech-2-mingw32-64.zip)

hiduino, 2019-09-10:
@facchinm, thank you, that version of bossac worked on Windows with the Nano Every!
I was able to update the MuxTo.