arduino/Arduino #9116: Not able to run YUN bridge on Omega2 board with arduino dock2

sumanjha090, 2019-08-01:
I had installed full python 2.7 and then installed YUN bridge and arduino dock 2 package.
Using below command i had enabled it
**uci set yunbridge.config.disabled='0'
uci commit yunbridge
reboot**

Then i had used arduino ide to upload console read program through wireless network. All done ok.

But SSH console not working, "telnet localhost 6571" throw error of refused connection.

I had gone through different solution on github and other forum. But got for arduino YUN board only. But nothing working for Omega2 board. How to fix it.

**Both the board (Omega2 and Arduino YUN) is having same chipset, RAM and ROM size. Is it possible to flash arduino YUN (U-Boot + kernel + rootfs-squashfs) on omega2. Please help me**

facchinm, 2019-08-02:
Hi @sumanjha090 ,
Omega2 board doesn't seem to contain a 32u4 microcontroller, which is the target for yun Bridge. Furthermore, `yunbridge` utility is not an Arduino program but comes from openwrt (I may be wrong on this).
Anyway, the right place to ask should be Omega2 forum/support.

closed 2019-08-02 by facchinm

sumanjha090, 2019-08-02:
Yes i know omega2 does not comes with 32u4, and i told i am using arduino dock 2 (which is having atmega328p--same arduino uno). And in YUN also linux and 32u4 communicate through UART only. So why can't be made with arduino UNO.

_Anyway, the right place to ask should be Omega2 forum/support._ I already asked them they are not able to help me, so i asked here. Is it not right place to ask. Its is linked to arduino.

Please don't ignore, help me

Thanks