arduino/Arduino #9121: Please add SparkFun's Qwiic Button/Switch Arduino Library

FischerMoseley, 2019-08-04:
At your earliest convenience, could you please add:

[https://github.com/sparkfun/SparkFun_Qwiic_Button_Arduino_Library](https://github.com/sparkfun/SparkFun_Qwiic_Button_Arduino_Library)

to the Arduino Library Manager? Thank you!

per1234, 2019-08-04:
Hi. The library.properties `name` value contains a disallowed character `/`.

From the Arduino library specification:
https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5:-Library-specification#libraryproperties-file-format
>- **name** - the name of the library. Library names must contain only basic letters (A-Z or a-z) and numbers (0-9), spaces ( ), underscores (_), dots (.) and dashes (-). It cannot start or end with a space, and also it cannot start with a number.

Please fix the `name` value and then either create a new tag/release or else redo the existing release.

Once that's done, please comment here so we'll know we can proceed.

FischerMoseley, 2019-08-04:
Hi per1234! Thanks for pointing that out, it should now be fixed.

cmaglie, 2019-08-21:
done

closed 2019-08-21 by cmaglie