arduino/Arduino #9122: Arduino Wifi Rev 2 - incompatibility between Tone and Servo

dcuartielles, 2019-08-05:
It is not possible to add a call to tone and, at the same time, invoke the servo. Seems like they are using the same timer. Simply try the following example to replicate the issue:

```
#include <Servo.h>

void setup() {
  // put your setup code here, to run once:
  pinMode(10, OUTPUT);
}

void loop() {
  // put your main code here, to run repeatedly:
  tone(10, 900);
}

```

This will produce the result: 

```
Tone.cpp.o (symbol from plugin): In function `tone(unsigned char, unsigned int, unsigned long)':
Se encontraron varias bibliotecas para "Servo.h"
(.text+0x0): multiple definition of `__vector_13'
libraries/Servo/megaavr/Servo.cpp.o (symbol from plugin):(.text+0x0): first defined here
Usado: /home/topeka/Arduino/libraries/Servo
collect2: error: ld returned 1 exit status
 No usado: /home/topeka/Development/arduino-1.8.9/libraries/Servo
exit status 1
Error compilando para la tarjeta Arduino Uno WiFi Rev2.

```

dcuartielles, 2019-08-05:
One possible solution would be changing the megaaver/ServoTimers.h to be using a different timer, e.g. 

```
//#define USE_TIMERB1        // interferes with PWM on pin 3
#define USE_TIMERB2        // interferes with PWM on pin 11
//#define USE_TIMERB0        // interferes with PWM on pin 6

```

But I am not sure whether this would provoke an issue somewhere else.

Note: this is needed for the EduIntro library to work

facchinm, 2019-08-06:
Fixed by merging https://github.com/arduino-libraries/Servo/pull/30

closed 2019-08-06 by facchinm