arduino/Arduino #9133: [Library Manager] Issue with Release

madichodikov, 2019-08-09:
Hi,

I recently created a new Release for the Arduino library manager, you can find the release here: 
https://github.com/sparkfun/SparkFun_TMP117_Arduino_Library/releases/tag/v1.1.0
Although, in the Arduino Library Manager, it has not updated yet to the correct version number. Is there something that I should do otherwise? I changed the library.properties file to match the version number, and I also changed the location but updated that in the initial issue to have the correct link, which can also be found here- 
https://github.com/sparkfun/SparkFun_TMP117_Arduino_Library
Let me know, thanks!

Madison

per1234, 2019-08-09:
I confirm the problem.

I notice two things:
- Between the 1.0.0 and 1.0.1 releases, the library.properties `name` value was changed from `SparkFun High Precision Temperature Sensor TMP117 Qwiic` to `SparkFun High Precision Temperature Sensor TMP117 Arduino Library`
- The new name value exceeds Arduino's maximum folder length of 63 characters (the folder name the Arduino Library Manager installs libraries to is based on the `name` value).

However, my understanding is that the library is locked to the original `name` value in the Library Manager index unless a request is made to update it, so I wouldn't expect either of those things to cause a problem in this case. I'll leave it to the expert, @cmaglie to investigate this. If cmaglie will confirm that Library Manager does enforce a maximum `name` value length of 63 characters then I'll add that information to the Arduino library specification.

madichodikov, 2019-08-13:
Hi per1234,

I just changed the name in the library.properties file back to what it was originally. Thank you for the information on the character length. Do you think that it is good to be able to re-update in the Arduino Library Manager? Thanks again!

Madison

cmaglie, 2019-08-21:
The problem is that the tag 1.1.0 has been made before you changed the library name back to the original.

```
JOB 2384 - 2019/08/21 07:31:17 Checking out tag: v1.0.0
JOB 2384 - 2019/08/21 07:31:17 Release SparkFun High Precision Temperature Sensor TMP117 Qwiic:1.0.0 already loaded, skipping
JOB 2384 - 2019/08/21 07:31:17 Checking out tag: v1.1.0
JOB 2384 - 2019/08/21 07:31:17 Release SparkFun High Precision Temperature Sensor TMP117 Arduino Library:1.1.0 has wrong library name, should be SparkFun High Precision Temperature Sensor TMP117 Qwiic
```

The libraries indexer still see `SparkFun High Precision Temperature Sensor TMP117 Arduino Library`:

https://github.com/sparkfun/SparkFun_TMP117_Arduino_Library/blob/v1.1.0/library.properties

to fix this you should:
1) delete the tag `1.1.0`
2) re-do the tag `1.1.0` on the master branch

per1234, 2019-11-19:
@madichodikov I notice that, although the v1.1.0 tag was never redone, as suggested by cmaglie in the last reply, there have been three new releases since that time, all of which are available from Library Manager as expected.

I know that modifying tags after they have been pushed to a public repo is considered by some to be not best practices, and it seems unlikely that the v1.1.0 release being missing from the Library Manager index is a serious problem at this point. So can we close this issue as solved?

madichodikov, 2019-11-19:
Yes, I will close the issue now! Thanks again

closed 2019-11-19 by madichodikov