arduino/Arduino #9136: Cannot see Arduino on Mojave

Russell108, 2019-08-11:
Purchased new Genuine MkR 1010 Arduino
Down loaded software from Arduino followed instructions
Told in those instructions no driver needed
cannot see Arduino

MBP Apple Mac mini usb c & USB ports

per1234, 2019-08-11:
Try using a different USB cable. You might have a damaged, defective, or charge-only cable.

The MKR WiFi 1010 comes with its pins stuck in black anti-static foam. This foam is electrically conductive so you need to remove it before using the MKR WiFi 1010.

Please let us know if that resolves the problem.

Russell108, 2019-08-11:
Thanks for the quick reply
This does not resolve the issue
Yes I read the instructions- I removed MKR 1010 from the foam
I have 4 cables the current one is from a Bose loudspeaker which then sends audio via usb so I believe It should be  a data cable.

Thanks

Russell108, 2019-08-11:
Ps I get a steady green LED & a quickly flashing red led on the MKR 1010 .

I have also ordered another usb cable just in case

rocketkittens, 2020-01-09:
I have a similar problem. OSX sucks, Tim Cook sucks, i'm selling my macbook.

Russell108, 2020-01-09:
Is it OS X or arduino not following
The protocol apple have initiated to keep our computers safer.

Linux is awesome, arduino is awesome, apple is awesome 

But they are all systems and we have to follow the rules

As c programmers we should know that

PaulStoffregen, 2020-01-09:
Just yesterday Cristian committed changes for Apple's new notarization stuff.  https://github.com/arduino/Arduino/commit/ac8017572372f18917a585e1ee356533b3004c68  Hopefully a fully notarized 1.8.11 release will be coming soon.

I did some experiments a few weeks ago with notarizing.  If you're willing to try an unofficial build, I have one you could use to check if that's the cause of these problems.  But my guess, from the very limited info here, is it won't make any difference.  When there's a hardware problem like a charge-only USB cable, no amount of software can make that communicate!

Russell108, 2020-01-09:
> Just yesterday Cristian committed changes for Apple's new notarization stuff. ac80175 <https://github.com/arduino/Arduino/commit/ac8017572372f18917a585e1ee356533b3004c68> Hopefully a fully notarized 1.8.11 release will be coming soon.

So sorry

What has a usb cable got to do with installing the software

I have the standard version of arduino running well on my computer

Its just that I need to update the firmware on the Arduino Nano IOT 33 wifi chip
I was told by Sandeep Mistry the following

You'll have to download the hourly Arduino build: https://www.arduino.cc/en/Main/Software#hourly <https://www.arduino.cc/en/Main/Software#hourly>
So I did then I came across another thing 

Its just there are some serious bugs in the wifi firmware which seem to be fixed by updates
Please read the GitHub stuff for further details which I have posted there on the 
Request of your support department.

I have also been told the fixes have been ready for some time just waiting now for the official up date

So sorry to disappoint it has nothing to do with my usb cables

per1234, 2020-10-02:
From the forum thread:
https://forum.arduino.cc/index.php?topic=630917.msg4300364#msg4300364
>my board was faulty.
...
I sent it back to supplier for replacement this worked fine.

So I'm closing this as resolved.

closed 2020-10-02 by per1234