arduino/Arduino #9137: Please add TinyPICO Helper to Library Manager

UnexpectedMaker, 2019-08-11:
Hi Arduino folks,

Can you please add the TinyPICO Helper to the Library Manager.

It's a helper library for my TinyPICO ESP32 board - https://tinypico.com  

Repository:
https://github.com/UnexpectedMaker/tinypico-helper

Release:
https://github.com/UnexpectedMaker/tinypico-helper/releases/tag/v1.0

Thanks!

per1234, 2019-08-11:
Hi. In order to be included in the Arduino Library Manager index, your library must be located in the root of the repository. It is currently located under the `TinyPICO-Helper` folder, which is not allowed. In addition to allowing your library to be added to Library Manager, moving the library to the root of the repository will also allow users the option to easily install it via the Arduino IDE's **Sketch > Include Library > Add .ZIP Library** feature.

Please fix the repository folder structure and then either create a new release/tag (after updating the `version` value in library.properties accordingly) or else redo the existing release/tag. Once that's done, please comment here so we'll know we can proceed.

More information:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

UnexpectedMaker, 2019-08-11:
Hey @per1234 :)

I knew to do that... no idea why I didn't, sorry :(
It's been redone and released as v1.1

https://github.com/UnexpectedMaker/tinypico-helper/releases/tag/v1.1

Cheers :-)
Seon

UnexpectedMaker, 2019-08-16:
@per1234 Hi - how often do the libraries get merged? I remember the last library I added was merged really quickly.. like less than a few days. Is there a schedule? Or does it just happen when folks have time?

per1234, 2019-08-16:
There's no set schedule. It's just done as the developer has time for it. The requests are allowed to accumulate for a while and then all cleared out at the same time.

UnexpectedMaker, 2019-08-17:
Ok cool, thanks for the clarification! Have a great weekend :-)

UnexpectedMaker, 2019-08-18:
Please use the latest v1.2 release...
https://github.com/UnexpectedMaker/tinypico-helper/releases/tag/v1.2

Thanks :-)

cmaglie, 2019-08-21:
I've added the library to the indexer, but it seems you forget to to update the version to 1.2.0 in library.properites:

https://github.com/UnexpectedMaker/tinypico-helper/blob/v1.2/library.properties

to fix this:
1) remove the `v1.2` tag
2) update version in `library.properties` to 1.2.0
3) commit & push
4) re-create the `1.2.0` tag on `master` branch

once done the libraries indexer should pick the version 1.2.0 automatically within an hour.

closed 2019-08-21 by cmaglie