arduino/Arduino #9141: [Library Manager] Please update link and name repo library LoraWAN Beelan

sabas1080, 2019-08-13:
Please update name and link the library

https://github.com/BeelanMX/arduino-LoRaWAN

Thanks!

cmaglie, 2019-08-21:
We are not accepting anymore submissions for libraries that have the prefix "Arduino*" in the name because we would like to reserve this namespace for our own libraries.

You've submitted the library before we introduced this rule so in theory it should not apply to your case but, since you're changing the name anyway, may I propose to change it to something like `Beelan LoraWAN` instead of `Arduino LoraWAN`? This has two advantages:

- it's immediately clear that the library is made (or is a fork made) by `Beelan`
- `Arduino` is redundant: this is a library for Arduino (the ecosystem), there is no need to add the word `Arduino` in the name!

sabas1080, 2019-09-18:
@cmaglie thanks, ready changes

cmaglie, 2019-10-16:
Done, sorry for the long delay.

closed 2019-10-16 by cmaglie