arduino/Arduino #9144: Verification Failure

LongChap, 2019-08-15:
Sketch verification fails after a sketch save. Just editing the sketch and then undoing the edit, ie (reverting back to the original sketch file, the sketch verification succeeds. Success verification messages say using previously compiled (cached?) code in temp location.
Particularly a problem for Nano.
Using IDE 1.8.9 on Windows 10 Pro, ver1903.

per1234, 2019-08-15:
When you encounter an error you'll see a button on the right side of the orange bar in the Arduino IDE "Copy error messages". Click that button. [Paste](https://lifehacker.com/how-to-copy-cut-and-paste-for-beginners-5801525) the error in a message here using [code fencing](https://guides.github.com/features/mastering-markdown/).

LongChap, 2019-08-15:
Hi

 

Here is a copy of the Verification Error messages 

 

 

Arduino: 1.8.9 (Windows 10), Board: "Arduino Nano, ATmega328P (Old Bootloader)"

 

C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\Gerry Phillips\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\Gerry Phillips\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries F:\Users\Gerry Phillips\Documents\Arduino\libraries -fqbn=arduino:avr:nano:cpu=atmega328old -ide-version=10809 -build-path R:\Users\Gerry Phillips\Temp\arduino_build_652860 -warnings=none -build-cache R:\Users\Gerry Phillips\Temp\arduino_cache_940752 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.arduinoOTA.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.arduinoOTA-1.2.1.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avrdude.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avrdude-6.3.0-arduino14.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avr-gcc.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avr-gcc-5.4.0-atmel3.6.1-arduino2.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -verbose F:\Users\Gerry Phillips\Documents\Arduino\GPS Clock\GPS_Template_09_TEST_PLL\GPS_Template_09_TEST_PLL.ino

C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\Gerry Phillips\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\Gerry Phillips\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries F:\Users\Gerry Phillips\Documents\Arduino\libraries -fqbn=arduino:avr:nano:cpu=atmega328old -ide-version=10809 -build-path R:\Users\Gerry Phillips\Temp\arduino_build_652860 -warnings=none -build-cache R:\Users\Gerry Phillips\Temp\arduino_cache_940752 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.arduinoOTA.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.arduinoOTA-1.2.1.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avrdude.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avrdude-6.3.0-arduino14.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avr-gcc.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avr-gcc-5.4.0-atmel3.6.1-arduino2.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -verbose F:\Users\Gerry Phillips\Documents\Arduino\GPS Clock\GPS_Template_09_TEST_PLL\GPS_Template_09_TEST_PLL.ino

Using board 'nano' from platform in folder: C:\Program Files (x86)\Arduino\hardware\arduino\avr

Using core 'arduino' from platform in folder: C:\Program Files (x86)\Arduino\hardware\arduino\avr

Detecting libraries used...

"C:\\Program Files (x86)\\Arduino\\hardware\\tools\\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10809 -DARDUINO_AVR_NANO -DARDUINO_ARCH_AVR "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\cores\\arduino" "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\variants\\eightanaloginputs" "R:\\Users\\Gerry Phillips\\Temp\\arduino_build_652860\\sketch\\GPS_Template_09_TEST_PLL.ino.cpp" -o nul

"C:\\Program Files (x86)\\Arduino\\hardware\\tools\\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10809 -DARDUINO_AVR_NANO -DARDUINO_ARCH_AVR "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\cores\\arduino" "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\variants\\eightanaloginputs" "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\libraries\\Wire\\src" "R:\\Users\\Gerry Phillips\\Temp\\arduino_build_652860\\sketch\\GPS_Template_09_TEST_PLL.ino.cpp" -o nul

"C:\\Program Files (x86)\\Arduino\\hardware\\tools\\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10809 -DARDUINO_AVR_NANO -DARDUINO_ARCH_AVR "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\cores\\arduino" "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\variants\\eightanaloginputs" "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\libraries\\Wire\\src" "-IF:\\Users\\Gerry Phillips\\Documents\\Arduino\\libraries\\LiquidCrystal" "R:\\Users\\Gerry Phillips\\Temp\\arduino_build_652860\\sketch\\GPS_Template_09_TEST_PLL.ino.cpp" -o nul

"C:\\Program Files (x86)\\Arduino\\hardware\\tools\\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10809 -DARDUINO_AVR_NANO -DARDUINO_ARCH_AVR "- [... cut ...]
Using library LiquidCrystal in folder: F:\Users\Gerry Phillips\Documents\Arduino\libraries\LiquidCrystal (legacy)

Using library TimerOne-v9 in folder: F:\Users\Gerry Phillips\Documents\Arduino\libraries\TimerOne-v9 (legacy)

Using library EEPROM at version 2.0 in folder: C:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries\EEPROM 

Using library RCSwitch in folder: F:\Users\Gerry Phillips\Documents\Arduino\libraries\RCSwitch (legacy)

exit status 1

Error compiling for board Arduino Nano.

 

This email has been checked for viruses by Avast antivirus software.
https://www.avast.com/antivirus

per1234, 2019-08-15:
Closing as duplicate of https://github.com/arduino/Arduino/issues/7949

This bug is specific to the 5.4.0-atmel3.6.1-arduino2 version of avr-gcc used by Arduino AVR Boards 1.6.22 and newer. 

#### Here's the workaround:
1. Tools > Board > Boards Manager
1. Wait for downloads to finish.
1. When you move the mouse pointer over "Arduino AVR Boards", you will see a "Select version" dropdown menu appear. Select "1.6.21".
1. Click "Install".
1. Wait for installation to finish.
1. Click "Close".

If you have **File > Preferences > Check for updates on startup** checked, the Arduino IDE may occasionally notify you that a new version of Arduino AVR Boards is available, you'll need to refrain from updating back to the new Arduino AVR Boards version, otherwise you'll be back to seeing the segmentation fault error again.

closed 2019-08-15 by per1234