arduino/Arduino #9147: [Library Manager] Please add fp64lib library to Library Manager

fp64lib, 2019-08-16:
Please add my library fp64lib for 64-bit floating point arithmetic, IEEE 754 compatible. Its repository can be found here:

             https://github.com/fp64lib/fp64lib

Documentation is available on http://fp64lib.org

Thank you and best regards,
Uwe

per1234, 2019-08-16:
Hi. There are a couple of problems that make your repository not compliant with the requirements for addition to the Arduino Library Manager index:

The library must be in the root of the repository. Your library is currently in the `source` subfolder of the repository. You will need to move it from there up to the root of the repository. I recommend removing the .zip file currently found in the root of the repository. This file serves no purpose as GitHub already provides automatically generated .zip files via **Clone or Download > Download ZIP** and the Releases page.

After doing that, you need to create a [release](https://help.github.com/articles/creating-releases/) or [tag](https://git-scm.com/docs/git-tag) that matches the `version` value in your library.properties file.

Once the issues above are fixed, please comment here to let us know we can proceed with adding your library to the Library Manager index.

---
For more information see:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

fp64lib, 2019-08-16:
Thanks for making me aware of the problems. I made the suggested changes and hope the repository is now compliant with the requirements for adding it to the Arduino Library Manager index.

cmaglie, 2019-08-21:
done

closed 2019-08-21 by cmaglie