arduino/Arduino #9148: Library Manager additions: Arduino-ThermistorLibrary

miguel5612, 2019-08-16:
Please, add the Arduino-thermistor library to the Arduino Library Manager. Here is the library repository:

Library URL: https://github.com/miguel5612/Arduino-ThermistorLibrary

Thank you!

miguel5612, 2019-08-16:
I was fixed an architecture problem, new version is 1.0.2

per1234, 2019-08-16:
Hi. Arduino recently started a new policy that 3rd party libraries with a library.properties `name` value starting with the word "Arduino" (or any capitalization thereof) will no longer be accepted for inclusion in the Arduino Library manager index. The reason is that Arduino wants to reserve these names for their own official libraries.

If you want your library to be added to the Library Manager index, you will need to change the `name` value in library properties to one that is compliant with the Arduino Library Specification:
https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5:-Library-specification#libraryproperties-file-format

Then update the `version` value in library.properties and create a new release/tag, or else redo the existing release/tag.

Note that this restriction only applies to the `name` value. You are not required to rename the repository or header files.

Once you have done this, please comment here so we'll know we can proceed.

More information:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

miguel5612, 2019-08-17:
Good afternoon, sir.
According to the request I have renamed the library removing the name Arduino- and I have also updated the lib.prop file updating the name and version. 
Thank you very much, I am waiting for your answer :D

cmaglie, 2019-08-21:
done

closed 2019-08-21 by cmaglie