arduino/Arduino #9152: Use of word Upload is inconsistent

bkelly13, 2019-08-18:
Please improve consistency in the Arduino IDE.  Consider changing the use of the word Upload within the IDE to Download.
The act or process of putting an application (sketch if you prefer) on the Arduino is being downloading the code. It is not Uploading.  If there is data on the Arduino and it is captured by the computer from the Arduino, that is an Upload operation.
Yes, uses can determine that when the IDE button is labeled Upload it really means Download.  That is an inconsistency that we just don't need.

matthijskooijman, 2019-08-18:
IMHO, this all depends on the perspective. Your computer uploads data *to* your Arduino, it is not the Arduino that is the active component here, it is your computer. You can certainly make an argument that Download is more intuitive, except:

 - It has been "Upload" since as long as I can remember, so changing the terminology will probably cause more confusion that it solves (especially with all kinds of online and third-party documentation that will still say "Upload").
 - AFAIK, a lot of other toolchains for embedded system use the term "upload" as well. This is not a really strong argument in the context of Arduino, but it plays along a bit.

So, if you ask me, this should not be changed (and this issue should probably be closed).

closed 2019-08-18 by per1234

bkelly13, 2019-08-20:
I expected that response.  You have the red pen.  It is a download and to call it upload is to use the antonym.

feikname, 2019-08-22:
It's both, you're uploading code to arduino and the arduino is downloading it. There's no such thing as upload without download. 

In my opinion, it makes total sense to use Upload since a user on a comptuer expects the computer itself to be the reference.

What could be changed to be more explicit, however, is the description when you hover the button to be "Upload to the board" or something of the sort
[image]