arduino/Arduino #9157: Ethernet.linkStatus() returning an incorrect status

bkelly13, 2019-08-20:
Arduino DUE, Arduino Ethernet Shield 2, IDE 1.8.8, windows 10
Application:  Web Server from David A Mellis and Tom Igoe,, The text of the code was copied from here
https://www.arduino.cc/en/Tutorial/WebServer

This appears to be a problem with function Ethernet.linkStatus() returning an incorrect status.  It declares that the Ethernet cable is not connected when it is.

I modified the app by putting code at the top of the loop to blink the LED_BUILTIN.  It does blink and the rate can be easily changed.  A solid indicator that the app is properly downloaded.

After a reset the serial port displays:  Ethernet cable is not connected.  

Setup:

My desktop is hard cabled to a D-Link switch which then connects to my cable modem. Again, a cable and not wireless. 
I started with the computer on port 1 of the D-Link and the Arduino on port 4.  
When the Ethernet cable is connected to the Arduino, check the lights at the Ethernet connection.
Looking at the connector from the perspective of the cable, the LED on the left is lit green solid.
The status light on the D-Link switch for that port is on.
The Arduino LED on the right side of the cable blinks amber irregularly.  
Disconnect the Ethernet cable and both go out.

Get another Ethernet cable, same results.

Move the desktop computer connection to port 4, I still have access to the internet.  Port 4 works, so does port 1.
Swap the cable that was between the Arduino and the D-Link with the cable between the D-Link and cable modem.  The desktop continues to have internet connectivity.  Now I have tried three cables and all three are good.

Replace the Arduino DUE and Ethernet shield with another sitting on my desk just for this reason.
Same Results.
Plug in a 12 Volt power supply so that the Arduino is now powered from the USB.
Same Results.

My project needs two Arduinos with Ethernet in separate locations, plus one more set for development.  Open the box and fire up the third set of Arduino and Ethernet shield.
Same results.

A ping from my computer does not solicit a reply.

From my perspective, the function Ethernet.linkStatus() is beyond my access and control.  Unless I really dig deep into this thing. 

What might I try now?

per1234, 2019-08-25:
Hi @bkelly13. Please move this to the issue tracker for the Ethernet library:
https://github.com/arduino-libraries/ethernet/issues

bkelly13, 2019-08-26:
I am using Chrome and Windows 10.  I don't see the categories and don't know how to move this to the Ethernet Library.  The only way I can find this thread is via a link I saved to it.
Would it help to re-install the Arduino.  IDE.  Since I have three complete CPU and ethernet shield setups all behaving the same, I might try that.

per1234, 2019-08-26:
Moved to https://github.com/arduino-libraries/Ethernet/issues/111

closed 2019-08-26 by per1234

bkelly13, 2019-08-30:
Problem found.  The Arduino IP was 192.168.1.177 while the router was set to 192.168.0.1.  Changed the Arduino to 192.168.0.177 at it began responding to the browser.
That was easy.  Once I knew what to change.
Thank you for your time.