arduino/Arduino #9159: Missing resistor - R26 on MKR WiFi 1010 boards

kierenj, 2019-08-20:
Hi - as a part of looking into setting the BQ24195L's OTG pin high to enable 5V boost operation, I've been poking around the board.

I have two boards, and on both, R26 is completely absent.

Can anyone else confirm? Is there any way to get an official word on why that is / what that means? At the same time it would be great to hear about what the deal is with controlling the OTG pin on the power management IC, this seems potentially relevant?

pnndra, 2019-08-20:
Hi r26 is supposed to be populated. Please send photos of the boards so that we can double check what may have gone wrong. Please let me also know where and when you purchased them so that we chan eventually track the batch that has this issue.

Thanks,

Dario

kierenj, 2019-08-21:
Thanks, my iPhone is having trouble focusing but I have a decent DSLR with macro lens and will get photos together this evening. I got them from Amazon.co.uk, sold by what looks like the official Arudino store: https://www.amazon.co.uk/sp?_encoding=UTF8&asin=&isAmazonFulfilled=1&isCBA=&marketplaceID=A1F83G8C2ARO7P&orderID=203-9578526-4504347&seller=A1LKZYWRVOF5T2&tab=&vasStoreID=

I'm not sure if it's a result of the code linked in #444 (in the core) using this as an output, or the board missing R26 here, but I determined the FET (Q3) on my first board was broken. That had no negative effect other than I believe it stopped me from toggling the OTG pin, but that's a discussion I was hoping to wrap up in #444.

Either way, I desoldered the FET, the OTG pin was left floating, the IC saw it as "high", and enabled boost mode and the 5V rail sprang to life without a battery. So one of my boards is minus a FET as well as a resistor.

(I haven't powered up my new board for fear of blowing that FET too, possibly by virtue of R26 missing or the potential bug linked in #444)

kierenj, 2019-08-22:
From my iPhone in the end after all.

Top board = I've desoldered the MOSFET Q3, which was broken/blown. Possibly caused by missing resistor?

Bottom board = new, straight from the box, unused

[image]

I would be so incredibly grateful to understand some answers around https://github.com/arduino/ArduinoCore-samd/issues/444 :)

kierenj, 2019-08-30:
Hi - indeed from the other thread it looks like my use case should work, if only I had R26 in place.

How can I return my boards, and most importantly, how can I by boards where this is populated?

Thank you! :)

Rocketct, 2019-08-30:
you have to take a little bit patience we to check in the warehouse if there are fault one and after we can send the good one, you can simply email the support@arduino.cc

kierenj, 2019-09-09:
I have my 2 replacements now (arrived with R26). I ordered a separate one from the Arudino website store directly, but I'll need to collect from the post office tomorrow.

The shipping costs from the official store are quite high. I'd like to be able to order from the Arudino Amazon UK store, but that's where my faulty ones were from. Do you know how widespread the problem was, if I can order from there again? Are there any discounts on shipping available from the official web store?

kierenj, 2019-09-16:
I ordered another board straight from the Arudino store, and it too has R26 missing.

On the forums, someone else said "both my boards have this missing too".

I've followed up with my support ticket too - where can I order boards with R26 present?

Thank you!