arduino/Arduino #9166: IDE becomes unresponsive after closing update dialog

joew46167, 2019-08-23:
The code update from pull #9164 can cause the IDE to become unresponsive.  This is only an issue when the use accessibility features option is enabled.

When the dialog box to update boards and/or libraries appears, if accessibility features are enabled and the user uses the close button, the main IDE becomes somehow impossible to access.  You can use modified keys for menu access or other hot key functions.  But it appears that you can not get into the editor and type there.

I think this is because the dialig box grabs focus?  I will need to look into this further.  

(Sorry Martino for a bad update.)

facchinm, 2019-08-23:
Don't worry, we have time to fix :wink: I'm adding the proper tags

per1234, 2021-04-03:
Closing as fixed by https://github.com/arduino/Arduino/commit/b9c3f008b0c3ff0da0804cf966087db3725d3570

closed 2021-04-03 by per1234