arduino/Arduino #9170: Arduino 1.8.9 high dpi issue on Linux

jpeterse, 2019-08-23:
Updated to 1.8.9 and menu's are not scaled with the Interface scaling setting.
The text in the editor is scaling, but not the menus.
Setting the GDL_SCALE environment variable doesn't work either.

Here's a screenshot comparing the menus from kwrite with arduino editor.
[image]

per1234, 2019-08-25:
Possibly related:
- https://github.com/arduino/Arduino/issues/5814
- https://github.com/arduino/Arduino/issues/6472

stapelberg, 2019-09-07:
As a shot into the dark: check your `gtk-font-name` value. I had `gtk-font-name "Bitstream Vera Sans 8"` in my `~/.gtkrc-2.0`. Changing font size from 8 to 12 worked for me.

jpeterse, 2019-09-07:
mine is currently:
gtk-font-name="Noto Sans Regular 10"
It looks like the menu's are driven by the font setting in the ~/.gtkrc-2.0 file .
The menu is not changing when the interface scaling is set in preferences.

Issue is, that Arduino IDE is the only application that doesn't display correctly on my PC.
So I don't want to change a global setting, to resolve the issue in a single application.

eminfedar, 2019-10-08:
Pardus 19.0 GNU/Linux: (with a screen has 96 dpi), everything is huge except the editor.
[image]

s-light, 2020-02-07:
same problem - menu font size does not scale..
as [discussed here](https://github.com/arduino/Arduino/pull/4376#issuecomment-505106291) i also tried  `-Dsun.java2d.uiScale=4` and `-Dglass.gtk.uiScale=400%` but this did not solve it for me.

the workaround by @stapelberg with setting the font-size in `~/.gtkrc-2.0` did work :-) :+1: 

so i [researched a bit](https://askubuntu.com/a/124460/207905) and to only set this for arduino i did the following:
- create a new file `arduino.gtkrc-2.0.conf` in my arduino folder
- copied & modified the content form the orig `.gtkrc-2.0` file to this: 
```
# This only overwrites the Font Settings for Menus to get readable.. :-)
include "/home/stefan/.gtkrc-2.0"
gtk-font-name="Noto Sans Regular 22"

```

now you can call your arduino ide with something like this:
`GTK2_RC_FILES=./arduino.gtkrc-2.0.conf arduino-1.8.10/arduino`
(if you are in the directory with the new special gtkrc file)

full path for me is:
`GTK2_RC_FILES=/home/stefan/mydata/tools/arduino/arduino.gtkrc-2.0.conf /home/stefan/mydata/tools/arduino/arduino-1.8.10/arduino`
and i just used this in my application launcher entry as the command.

BackMountainDevil, 2021-06-04:
` ~/.gtkrc-2.0`  not work for Arch KDE + arduino 1.8.5