arduino/Arduino #9178: [Library Manager] Please add the EnergyMeter library

elC0mpa, 2019-08-27:
I have created an EnergyMeter library for Arduino and wonder if you see could include it in the IDE library manager.
This library is used to monitor energy meters. It uses callbacks to trigger actions each time the meter measures a certain kind of kilowatts-hour. By using this library, energy meters can be polled (less efficient approach) or can be interfaced through external interrupts (most efficient approach).

The project and its documentation are available at https://github.com/elC0mpa/EnergyMeter/

Regards :)

elC0mpa, 2019-09-01:
Hello @per1234, I would like to know if everything is right because i have been waiting for you to add my library since five days. I thought maybe the .properties file was wrong. Please answer me, I would like to know your opinion

per1234, 2019-09-01:
Everything is fine now.

Your library will not be available for installation via Library Manager until it has been added to the index. cmaglie occasionally comes through to clear all the outstanding Library Manager inclusion requests and closes those issues when that is done. You will receive a notification from GitHub of this and your library will be available in Library Manager shortly after that. So we'll get to it soon, please be patient.

cmaglie, 2019-09-14:
done, sorry for the long delay.

closed 2019-09-14 by cmaglie