arduino/Arduino #9180: [website] error specs Arduino Micro

StefanT83, 2019-08-28:
I believe there is an error on the "Tech Specs" link of the Arduino Micro, namely on these two websites:
https://store.arduino.cc/arduino-micro
https://store.arduino.cc/arduino-micro-without-headers

Specifically, the indication "DC Current per I/O Pin | 20 mA" is not correct: it should be 40mA. This information can be found on the datasheet of ATmega32U4, Section 29.1, page 383.
Interestingly, the same information appears correctly indicated on the "Tech Specs" of the Arduino Leonardo, which uses the same chipset as the Arduino Micro:
https://store.arduino.cc/leonardo

per1234, 2019-08-28:
Note that 40 mA is the absolute maximum rating. As the datasheet says:
>This is a stress rating only
and functional operation of the device at these
or other conditions beyond those indicated in
the operational sections of this specification is
not implied. Exposure to absolute maximum rat-
ing conditions for extended periods may affect
device reliability.

Stating only:
>
DC Current per I/O Pin | 40 mA
-- | --

Will give the users the impression that they are safe to go all the way up to 40 mA.

I notice that note 3 under table 29-1 says:
> 3. Although each I/O port can sink more than the test conditions (20mA at V CC = 5V, 10mA at V CC = 3V) under steady state
conditions (non-transient), the following must be observed:
ATmega16U4/ATmega32U4:
1.)The sum of all IOL, for ports A0-A7, G2, C4-C7 should not exceed 100mA.
2.)The sum of all IOL, for ports C0-C3, G0-G1, D0-D7 should not exceed 100mA.
3.)The sum of all IOL, for ports G3-G5, B0-B7, E0-E7 should not exceed 100mA.
4.)The sum of all IOL, for ports F0-F7 should not exceed 100mA.
>
>If IOL exceeds the test condition, VOL may exceed the related specification. Pins are not guaranteed to sink current
greater than the listed test condition.

So I guess that's where the 20 mA number comes from? I've always thought it was safe to go at least to 30 mA, but maybe 20 mA is the safe maximum.

So it seems to me that it's really the Leonardo's product page that has the incorrect number.

kevinjwalters, 2025-01-19:
**_UNRELATED COMMENT_** 

For #11470 which I cannot comment on as it's been locked, there are at least two places this hasn't been updated for the Leonardo board:

- https://docs.arduino.cc/resources/pinouts/A000057-full-pinout.pdf
- https://docs.arduino.cc/language-reference/en/functions/communication/SPI/#description (noted by @gentilkiwi)

I think @FunzioneFabio had worked on rolling this change out.

gentilkiwi, 2025-01-20:
> **_UNRELATED COMMENT_**
> 
> For [#11470](https://github.com/arduino/Arduino/issues/11470) which I cannot comment on as it's been locked, there are at least two places this hasn't been updated for the Leonardo board:
> 
> * https://docs.arduino.cc/resources/pinouts/A000057-full-pinout.pdf
> * https://docs.arduino.cc/language-reference/en/functions/communication/SPI/#description (noted by [@gentilkiwi](https://github.com/gentilkiwi))
> 
> I think either [@FunzioneFabio](https://github.com/FunzioneFabio) or [@kengdahl](https://github.com/kengdahl) had worked on rolling this change out.

Unfortunately it seems they don't want to deal with it :(
https://github.com/BdF-LabSec/kpn532/issues/1#issuecomment-2563517164

kengdahl, 2025-01-28:
@per1234 I don't think I can be of assist on this one.