arduino/Arduino #9185: Button problem in Preferences

joew46167, 2019-08-29:
From Ken

> Button problem in Preferences
>
> The new window button is miss labeled for accessibility. When you pres the new window button it opens the add boards edit box into a new window. Of course until I push the button I have no idea what new window button does. Maybe it should be changed. For example
>
> You should be able to read the edit box in the preferences but if you want to add boards you have to push the add button to open the dialog. As it is now you can add it in the menu or after you press the button. What do you think?
>
> I am confused why there is both.

My take on this

Even as a sighted user that button isn't very intuitive.

Here's my take on it. In the Additional Boards Manager URLs edit box you can add text to specify a single URL to be used by the board manager. But if you want to add multiple URLs you need to use the dialog box so that you can enter each URL on a single line. Makes sense, I guess.

But the button is awful! It is an image button that opens up to the dialog box (I'd call it a dialog box, not a window). From the image it is not clear to me what the button does. And the screen reader description isn't any clearer.

I agree the button needs to be fixed for the screen reader. But it needs to be fixed for the sighted user, too!

per1234, 2019-08-29:
>But if you want to add multiple URLs you need to use the dialog box so that you can enter each URL on a single line.

You do have the option of adding multiple URLs to that field separated by commas.

>I agree the button needs to be fixed for the screen reader. But it needs to be fixed for the sighted user, too!

I have noticed that when I am providing support I find it easier to just tell people to use the edit box with comma separators instead of trying to explain the process of clicking that button.

joew46167, 2019-08-29:
I went and looked at the Preferences dialog again.  

First off, that button is the only graphical button on the page.  It's not the only button.  I'm inclined to make it also be a text button with the text "Multiple URLs" and the screen reader label (if I can) say multiple board manager URLs. 

Next, there is no way to know that the edit field is a comma separated list rather than a single URL. I just learned that today! That's true for both people who are sighted and people who are vision impaired.  It already has a very long label though, so adding use commas to separate URLs will make that even worse.

I'm not a designer, so I'm not sure how best to approach this second point.

*side note:  I just saw the tool tip that says you can enter multiple URLs separated by columns - I'd actually never used that text entry with a mouse before, always by tabbing there.