arduino/Arduino #9188: Comply with XDG

A6GibKm, 2019-08-29:
Currently Arduino IDE stores packages and others in `$HOME/.arduino15/`, this should be done in `$XDG_DATA_HOME/arduino` and preferences should be located in `$XDG_CONFIG_HOME/arduino`. The use of `$HOME/.arduino15` should be used only as a fallback.

Related to #8659.

per1234, 2019-08-30:
Closing as duplicate of https://github.com/arduino/Arduino/issues/3915

closed 2019-08-30 by per1234

cvoges12, 2020-08-27:
Stop closing it. This is a change you guys should make and have been incredibly stubborn for not moving through with a simple request that has been made multiple times already, and closed for literally **no reason**. I'd be happy to implement it if you @per1234 would stop closing it. This is not in the spirit of open source

This is retarded and embarassing to the project. See: https://wiki.archlinux.org/index.php/XDG_Base_Directory for more information. @A6GibKm and others have totally been in the right. Most major applications have made the move already to the standard.

Stop closing and locking the issue. This is dumb and childish

per1234, 2020-08-27:
I will continue to close duplicates. We only allow one issue for each feature request or bug report. The reason is that this allows easily tracking the issue and confines all the information and discussion on the topic to a single location.

I would argue that making a mess of the issue tracker of an open source software project, wasting the time of the maintainers, developers, users, and followers of that project is what's not in the spirit of open source.

If you have original, productive input to provide, you are welcome to do so on the original issue: https://github.com/arduino/Arduino/issues/3915. However, I'll warn you now that using insults will render your comment unproductive and we have zero tolerance for the type of offensive language you used here.