arduino/Arduino #9192: I belive you got a bug...

offcircuit, 2019-09-02:
Hi,

I dont know if this is a bug

SoftwareSerial.h

line 113

#ifndef ESP32
  **ssize_t** m_swsInstsIdx = -1; <---- this should be size_t m_swsInstsIdx 
#endif

Thanks,
Helder Rodrigues

per1234, 2019-09-03:
Each architecture uses its own version of the SoftwareSerial library so there are multiple versions of the library. Please specify which SoftwareSerial library you found this bug in.

offcircuit, 2019-09-03:
Hi again,

I found this at

**file**
C:\Users\offcircuit\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.5.2\libraries\SoftwareSerial\src/SoftwareSerial.h

**dowloaded from**

http://arduino.esp8266.com/stable/package_esp8266com_index.json

**error**

In file included from sketch\Nextion.h:7:0,

                 from sketch\Nextion.cpp:1:

C:\Users\offcircuit\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.5.2\libraries\SoftwareSerial\src/SoftwareSerial.h:113:2: error: 'ssize_t' does not name a type

  ssize_t m_swsInstsIdx = -1;

  ^

exit status 1
Erro ao compilar para a placa LOLIN(WEMOS) D1 R2 & mini.

per1234, 2019-09-03:
The ESP8266 core for Arduino is a 3rd party project which Arduino has no direct involvement in. We only accept bug reports for official Arduino software here. You'll need to report this to the ESP8266 people:
https://github.com/esp8266/Arduino

closed 2019-09-03 by per1234