arduino/Arduino #9196: We editor browser add in malware XIAZAI

Insubson, 2019-09-04:
Did anyone else get browser editor Xiazai malware after installing the Arduino web editor browser add in? I can't definitively say that it is the cause but considering that this is the only software I've installed in the last year I think this is the source. 

This laptop is only used for coding and has 32GB Storage and 2GB RAM so has trouble with web access even. So only use for arduino.

Has anyone else had this problem?

per1234, 2019-09-04:
Yes. There are multiple reports of this on the Arduino forum and Arduino is working on this:
https://forum.arduino.cc/index.php?topic=634561.msg4296455#msg4296455
>We've filed a case to Microsoft as we believe it's a false positive. I hope they will get back to us soon and remove it from their AV signatures.

Please follow that forum thread for updates.

According to https://github.com/arduino/arduino-create/issues/1#issuecomment-351344954, the [Create > Editor section of the Arduino Forum](http://forum.arduino.cc/index.php?board=101.0) is the appropriate place to submit bug reports or feature requests related to the Arduino Web Editor. Since we already have a report there, I'm going to close this issue.

closed 2019-09-04 by per1234

Insubson, 2019-09-04:
Oh thank you. Your response was fast. That's great news. Much appreciated.